carla-simulator/carla-autoware #5: ImportError: No module named ackermann_msgs.msg

jan-reich, 2019-02-01:
I built the carla-autoware-bridge as described in the install notes. 
When executing 
`roslaunch carla_autoware_bridge carla_autoware_bridge.launch`
the vehiclecmd_to_ackermanndrive node is dying with the following error message:
`Traceback (most recent call last):
  File "/home/reich/carla-autoware/catkin_ws/src/carla_autoware_bridge/src/vehiclecmd_to_ackermanndrive", line 4, in <module>
    from ackermann_msgs.msg import AckermannDrive
ImportError: No module named ackermann_msgs.msg
[vehiclecmd_to_ackermanndrive-6] process has died [pid 30370, exit code 1, cmd /home/reich/carla-autoware/catkin_ws/src/carla_autoware_bridge/src/vehiclecmd_to_ackermanndrive __name:=vehiclecmd_to_ackermanndrive __log:=/home/reich/.ros/log/ed39ca24-2679-11e9-8cb1-485b3976f12a/vehiclecmd_to_ackermanndrive-6.log].
log file: /home/reich/.ros/log/ed39ca24-2679-11e9-8cb1-485b3976f12a/vehiclecmd_to_ackermanndrive-6*.log`

As the normal ros-bridge is already dependent on the AckermannDrive message, I wonder, where this message is defined as I didn't find any .msg file for it.

Could someone help me in this regard?

fpasch, 2019-02-02:
Please make sure, that you have installed ros-kinetic-​ackermann-msgs.

jan-reich, 2019-02-02:
Thank you very much :-)

closed 2019-02-02 by jan-reich

mrgransky, 2020-07-21:
`sudo apt-get update`
`sudo apt-get install ros-kinetic-​ackermann-msgs`
should fix the issue!