carla-simulator/carla-autoware #20: Slow in manual driving mode

wzhang1, 2019-03-18:
It seems that in some town(for example town4), autopilot can't traverse the map quickly, so I tried to do creation with manual driving, for example, change the option in  line 3 in carla_autoware_bridge_capture_pcl_map.launch to do manual driving with mapping, 
    <param name="/carla/client/start_in_autopilot" value="True" /> 

The reaction in pygame is quite slow in this case, is that because of  my hardware to slow?
e.g. recent intel processors and nvidia card, fps in server = 10, and client fps >>10,

fpasch, 2019-03-19:
Which version of Carla + ros-bridge are you using?

wzhang1, 2019-03-19:
thanks, I followed the instruction of Redme in this repo,
Carla == 0.9.4
ros-bridge @ f86267d

fpasch, 2019-03-29:
With #27 , the overall setup is changing. There is a manual_control node, that receives its image-data via ROS (and not from the Carla Client itself).

closed 2019-04-01 by fpasch