carla-simulator/carla-autoware #42: Fail to build Carla Autoware Bridge

sheepwxy, 2019-04-26:
Hello!
when I do the roslaunch $CARLA_AUTOWARE_ROOT/devel.launch,it goes wrong and the following error occurred`ResourceNotFound: points_preprocessor
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/wuxuyang/carla-autoware/catkin_ws/src
ROS path [2]=/home/wuxuyang/catkin_ws/src
ROS path [3]=/opt/ros/kinetic/share
`
Can someone help me?

fpasch, 2019-04-29:
You're missing the Autoware catkin workspace?
Please follow exactly the steps to set up your shell.

AravindChandradoss, 2019-05-22:
```
ResourceNotFound: points_preprocessor
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/autoware/carla-autoware/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

```
@sheepwxy @fpasch I am having the same issue. How did you solve? I exactly followed the instruction.
I build Autoware using colcon.
I launched Carla from Unreal Editor. 

Am I missing anything?

Thanks

yeseraccenture, 2019-05-23:
@sheepwxy @fpasch @AravindChandradoss  I have exact the same problem:

`ResourceNotFound: points_preprocessor
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/autoware/carla-autoware/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
`
Any idea about how to solve?

fpasch, 2019-05-23:
source <path-to-autoware>/ros/install/setup.bash

yeseraccenture, 2019-05-23:
`   12  export CARLA_AUTOWARE_ROOT=/home/autoware/carla-autoware/autoware_launch
   13  export CARLA_MAPS_PATH=/home/autoware/carla-autoware/autoware_data/maps
   14  source /home/autoware/Autoware/ros/install/setup.bash 
   15  source devel/setup.bash 
   16  roslaunch \$CARLA_AUTOWARE_ROOT/devel.launch 
   17  roslaunch ../autoware_launch/devel.launch `

this is my history . Exact order is above. Is there any problem?  I source autoware before carla-autoware. Any idea?

AravindChandradoss, 2019-05-23:
@fpasch we cannot source multiple workspace like that. 
```
source <path-to-autoware>/ros/install/setup.bash
source $CARLA_AUTOWARE_ROOT/../catkin_ws/devel/setup.bash
```
when we source setup.bash, it will overlay all the existing sources. (Thats the reason we werenot able find points_preprocess). We have to rebuild either autoware or carla. 

Follow the steps to have both the workspace in a single setup.bash.
1) remove build and devel in catkin_ws in carla-autoware folder. 
`sudo rm -R devel build`
2) source the autoware ros setup.bash here **(Dont forget this step)**
`source <path-to-autoware>/ros/install/setup.bash`
3) do catkin_make in catkin-ws folder
`catkin_make`

Now follow the steps as in documentation.

Sanity check:
after sourcing `source $CARLA_AUTOWARE_ROOT/../catkin_ws/devel/setup.bash`
check Ros Path
`echo $ROS_PACKAGE_PATH`
you will see many source path. (Many!)

yeseraccenture, 2019-05-24:
Thanks both of you for the answer. I did what @AravindChandradoss wrote but I couldn't see the MANY paths, only two.
`cd carla-autoware/catkin_ws/
 sudo rm -R devel/ build/
 export AUTOWARE_ROOT=/mnt/HDD/carla_ue4_space/autoware
 source /mnt/HDD/carla_ue4_space/autoware/ros/install/setup.bash 
 source /opt/ros/kinetic/setup.bash 
 catkin_make
 source /mnt/HDD/carla_ue4_space/carla-autoware/autoware_launch/../catkin_ws/devel/setup.bash 
 echo $ROS_PACKAGE_PATH `

output is:
`/mnt/HDD/carla_ue4_space/carla-autoware/catkin_ws/src:/opt/ros/kinetic/share`

Still thinking about what I am doing wrong. I am beginner to this and your help is very helpful for me and people who wants to use the bridge.  
@AravindChandradoss , which steps in documentation I need to follow? Running terminal 2 part from the beginning? I can give necessary information if asked. Thanks in advance

AravindChandradoss, 2019-05-24:
I think you dont need to source /opt/ros/kinetic/setup.bash (mostly they will be in .bashrc by default).
If you dont mind, share your `echo $ROS_PACKAGE_PATH` output after sourcing `source <path-to-autoware>/ros/install/setup.bash`.
If you see many source paths. just do the` catkin_make` and follow the documentation from [here](https://github.com/carla-simulator/carla-autoware#run) 

let me know if you any runtime error as in [issue](https://github.com/carla-simulator/carla-autoware/issues/50)
For now, I am waiting for @fpasch  or someone from carla to help in that issue.

yeserac, 2019-05-27:
This is the output after sourcing autoware and catkin_make:
`/mnt/HDD/carla_ue4_space/autoware/ros/install/ymc/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/xsens_driver/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/lattice_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/waypoint_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/waypoint_maker/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/way_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/trafficlight_recognizer/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_utilities/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_simulation_package/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_local_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_global_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/lidar_kf_contour_track/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_ros_helpers/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/lane_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/ff_waypoint_follower/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/dp_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/waypoint_follower/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vlg22c_cam/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_ssd_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_segment_enet_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_lane_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_dpm_ttic_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_darknet_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vision_beyond_track/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vehicle_socket/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vehicle_model/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vehicle_gazebo_simulation_launcher/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vehicle_gazebo_simulation_interface/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vehicle_description/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_simu/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_planner/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/op_utility/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/lidar_euclidean_cluster_detect/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vector_map_server/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/road_occupancy_processor/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/costmap_generator/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/object_map/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/naive_motion_predict/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/map_file/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/libvectormap/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/imm_ukf_pda_track/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/decision_maker/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vector_map/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vector_map_msgs/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/vectacam/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/udon_socket/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/twist_generator/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/tablet_socket/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/runtime_manager/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/mqtt_socket/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/tablet_socket_msgs/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/state_machine_lib/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sound_player/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_lms5xx/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_ldmrs_laser/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_ldmrs_tools/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_ldmrs_driver/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_ldmrs_msgs/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/sick_ldmrs_description/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/points2image/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/rosinterface/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/rosbag_controller/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/roi_object_filter/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/range_vision_fusion/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/pos_db/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/points_preprocessor/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/points_downsampler/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/pixel_cloud_fusion/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/lidar_localizer/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/pcl_omp_registration/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/pc2_downsampler/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/oculus_socket/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/obj_db/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/nmea_navsat/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/ndt_tku/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/ndt_gpu/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/ndt_cpu/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/multi_lidar_calibrator/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/microstrain_driver/share:/mnt/HDD/carla_ue4_space/autowar [... cut ...] D/carla_ue4_space/autoware/ros/install/autoware_driveworks_gmsl_interface/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/autoware_config_msgs/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/autoware_can_msgs/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/autoware_build_flags/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/autoware_bag_tools/share:/mnt/HDD/carla_ue4_space/autoware/ros/install/adi_driver/share:/opt/ros/kinetic/share`

The unused args problem comes when I proceed.

AravindChandradoss, 2019-05-28:
@yeserac were you able to successfully launch the bridge with the runtime error that I mentioned? 
what was your problem? cant get you. _Unused args?_

yeserac, 2019-05-29:
I get unused args:
`unused args [enablePlannerDynamicSwitch] for include of [/mnt/HDD/carla_ue4_space/autoware/ros/install/waypoint_planner/share/waypoint_planner/launch/velocity_set.launch] The traceback for the exception was written to the log file`
I am not sure it is succesful launching or not. I didn't get your error though. If I repeat my steps, I still get the above error.

ntutangyun, 2019-05-30:
> I get unused args:
> `unused args [enablePlannerDynamicSwitch] for include of [/mnt/HDD/carla_ue4_space/autoware/ros/install/waypoint_planner/share/waypoint_planner/launch/velocity_set.launch] The [...]

having the same issue.

AravindChandradoss, 2019-05-30:
Can you guys share your steps, so that I can try to reproduce the same.
btw, did you guys used the docker?

AravindChandradoss, 2019-05-30:
The issue can be fixed by removing parameter while launching.
**enablePlannerDynamicSwitch** was remove from Autoware as in [here](https://github.com/autowarefoundation/autoware/commit/9f7bf977125af469fad902ec42c8073e9ac7072a).

So, remove _enablePlannerDynamicSwitch_ from launch files. 
Now, the bridge is working.

ntutangyun, 2019-05-31:
> The issue can be fixed by removing parameter while launching.
> **enablePlannerDynamicSwitch** was removed from Autoware as in [here](https://github.com/autowarefoundation/autoware/commit/9f7bf977125af469fad902ec42c8073e9ac7072a).
> 
> So, remove _enablePlannerDynamicSwitch_ from launch files.
> Now, the bridge is working.

Hi @AravindChandradoss , I'm not sure how it is fixed by removing parameters while laucnhing. 

I checked all the files changed in the commit link you mentioned, all the files under Autoware folder doesn't contain the so called "enablePlannerDynamicSwitch", meaning this is already removed I supposed.

But I'm still getting this error: 

`[roslaunch][INFO] 2019-05-31 13:39:49,273: starting in server mode
[roslaunch.parent][INFO] 2019-05-31 13:39:49,273: starting roslaunch parent run
[roslaunch][INFO] 2019-05-31 13:39:49,274: loading roscore config file /opt/ros/kinetic/etc/ros/roscore.xml
[roslaunch][INFO] 2019-05-31 13:39:49,700: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2019-05-31 13:39:49,700: loading config file /home/tangyun/tools/carla-autoware/autoware_launch/devel.launch
[roslaunch][INFO] 2019-05-31 13:39:49,767: Added node of type [carla_manual_control/carla_manual_control.py] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,790: Added node of type [carla_ros_bridge/client.py] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,795: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,796: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,796: Added node of type [carla_ego_vehicle/carla_ego_vehicle.py] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,802: Added node of type [carla_waypoint_publisher/carla_waypoint_publisher.py] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,806: Added node of type [carla_points_map_loader/carla_points_map_loader] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,816: Added node of type [tf/static_transform_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,816: Added node of type [tf/static_transform_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [tf/static_transform_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [tf/static_transform_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [tf/static_transform_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,817: Added node of type [carla_autoware_bridge/odometry_to_posestamped] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,818: Added node of type [carla_autoware_bridge/vehiclecmd_to_ackermanndrive] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,818: Added node of type [carla_autoware_bridge/carla_to_autoware_vehicle_status] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,818: Added node of type [carla_autoware_bridge/carla_to_autoware_waypoints] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,829: Added node of type [carla_ackermann_control/carla_ackermann_control_node.py] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,834: Added node of type [points_preprocessor/ray_ground_filter] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,851: Added node of type [joint_state_publisher/joint_state_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,852: Added node of type [robot_state_publisher/state_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,856: Added node of type [autoware_connector/can_odometry] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,860: Added node of type [points_downsampler/voxel_grid_filter] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,864: Added node of type [lidar_localizer/ndt_matching] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,869: Added node of type [lidar_euclidean_cluster_detect/lidar_euclidean_cluster_detect] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,870: Added node of type [detected_objects_visualizer/visualize_detected_objects] in namespace [/detection/lidar_detector/]
[roslaunch][INFO] 2019-05-31 13:39:49,875: Added node of type [autoware_connector/can_status_translator] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,875: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,875: Added node of type [topic_tools/relay] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,876: Added node of type [waypoint_maker/waypoint_marker_publisher] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,876: Added node of type [lane_planner/lane_rule] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,876: Added node of type [lane_planner/lane_stop] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,876: Added node of type [lane_planner/lane_select] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,881: Added node of type [costmap_generator/costmap_generator] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,884: Added node of type [waypoint_planner/astar_avoid] in namespace [/]
[roslaunch][INFO] 2019-05-31 13:39:49,887: Added node of type [waypoint_planner/velocity_set] in namespace [/]
[... cut ...]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 455, in load_config_default
    raise RLException(e)
RLException: unused args [enablePlannerDynamicSwitch] for include of [/home/tangyun/tools/Autoware/ros/install/waypoint_planner/share/waypoint_planner/launch/velocity_set.launch]
`

How to configure roslaunch to use python3.5 instead of python2.7 i wonder ??

Thank You

yeserac, 2019-05-31:
I did what @ntutangyun wrote and I got the same issue:
`[carla_manual_control_ego_vehicle-1] process has died [pid 9980, exit code 1, cmd /mnt/HDD/carla_ue4_space/carla-autoware/catkin_ws/src/ros-bridge/carla_manual_control/src/carla_manual_control/carla_manual_control.py __name:=carla_manual_control_ego_vehicle __log:=/home/acn/.ros/log/f85d343e-8382-11e9-8eca-c04a000396d4/carla_manual_control_ego_vehicle-1.log].
log file: /home/acn/.ros/log/f85d343e-8382-11e9-8eca-c04a000396d4/carla_manual_control_ego_vehicle-1*.log
[carla_ego_vehicle_ego_vehicle-5] process has died [pid 9985, exit code 1, cmd /mnt/HDD/carla_ue4_space/carla-autoware/catkin_ws/src/ros-bridge/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py __name:=carla_ego_vehicle_ego_vehicle __log:=/home/acn/.ros/log/f85d343e-8382-11e9-8eca-c04a000396d4/carla_ego_vehicle_ego_vehicle-5.log].
log file: /home/acn/.ros/log/f85d343e-8382-11e9-8eca-c04a000396d4/carla_ego_vehicle_ego_vehicle-5*.log
`

fpasch, 2019-06-06:
@AravindChandradoss: Could you check if the documentation is correct with #52

fpasch, 2019-06-06:
@ntutangyun Could you please review: #53

fpasch, 2019-06-06:
@ntutangyun @yeserac Are you still facing the "the ego vehicle process has died." issue?

yeserac, 2019-06-10:
I reviewed the changes and executed them exactly. I still have the error and I am copying the full output:

`acn@acn-16:/mnt/HDD/carla_ue4_space/carla-autoware$ roslaunch autoware_launch/devel.launch 
... logging to /home/acn/.ros/log/f8f97e2c-8b41-11e9-9c48-c04a000396d4/roslaunch-acn-16-5222.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://acn-16:33109/

SUMMARY
========

PARAMETERS
 * /astar_avoid/angle_goal_range: 6.0
 * /astar_avoid/avoid_start_velocity: 3.0
 * /astar_avoid/avoid_waypoints_velocity: 10.0
 * /astar_avoid/curve_weight: 1.2
 * /astar_avoid/distance_heuristic_weight: 1.0
 * /astar_avoid/enable_avoidance: False
 * /astar_avoid/lateral_goal_range: 0.5
 * /astar_avoid/longitudinal_goal_range: 2.0
 * /astar_avoid/minimum_turning_radius: 6.0
 * /astar_avoid/obstacle_threshold: 100
 * /astar_avoid/potential_weight: 10.0
 * /astar_avoid/replan_interval: 2.0
 * /astar_avoid/reverse_weight: 2.0
 * /astar_avoid/robot_base2back: 1.0
 * /astar_avoid/robot_length: 4.5
 * /astar_avoid/robot_width: 1.75
 * /astar_avoid/safety_waypoints_size: 100
 * /astar_avoid/search_waypoints_delta: 2
 * /astar_avoid/search_waypoints_size: 50
 * /astar_avoid/theta_size: 48
 * /astar_avoid/time_limit: 1000.0
 * /astar_avoid/update_rate: 10
 * /astar_avoid/use_back: False
 * /astar_avoid/use_potential_heuristic: True
 * /astar_avoid/use_wavefront_heuristic: False
 * /carla/ackermann_control/accel_Kd: 0.05
 * /carla/ackermann_control/accel_Ki: 0.0
 * /carla/ackermann_control/accel_Kp: 0.02
 * /carla/ackermann_control/min_accel: 1.0
 * /carla/ackermann_control/speed_Kd: 0.5
 * /carla/ackermann_control/speed_Ki: 0.0
 * /carla/ackermann_control/speed_Kp: 0.05
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/host: localhost
 * /carla/maps_path: /mnt/HDD/carla_eu...
 * /carla/port: 2000
 * /carla_ackermann_control_ego_vehicle/role_name: ego_vehicle
 * /carla_ego_vehicle_ego_vehicle/role_name: ego_vehicle
 * /carla_ego_vehicle_ego_vehicle/sensor_definition_file: /mnt/HDD/carla_ue...
 * /carla_ego_vehicle_ego_vehicle/spawn_point: 
 * /carla_ego_vehicle_ego_vehicle/vehicle_filter: vehicle.toyota.pr...
 * /carla_manual_control_ego_vehicle/role_name: ego_vehicle
 * /carla_waypoint_publisher/role_name: ego_vehicle
 * /costmap_generator/expand_polygon_size: 1.0
 * /costmap_generator/grid_length_x: 50
 * /costmap_generator/grid_length_y: 30
 * /costmap_generator/grid_max_value: 1.0
 * /costmap_generator/grid_min_value: 0.0
 * /costmap_generator/grid_position_x: 20
 * /costmap_generator/grid_position_y: 0
 * /costmap_generator/grid_resolution: 0.2
 * /costmap_generator/lidar_frame: velodyne
 * /costmap_generator/map_frame: map
 * /costmap_generator/maximum_lidar_height_thres: 0.3
 * /costmap_generator/minimum_lidar_height_thres: -2.2
 * /costmap_generator/size_of_expansion_kernel: 9
[... 246 lines cut ...]
[carla_ros_bridge-3] process has died [pid 5249, exit code 1, cmd /mnt/HDD/carla_ue4_space/carla-autoware/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/acn/.ros/log/f8f97e2c-8b41-11e9-9c48-c04a000396d4/carla_ros_bridge-3.log].
log file: /home/acn/.ros/log/f8f97e2c-8b41-11e9-9c48-c04a000396d4/carla_ros_bridge-3*.log
[ INFO] [1560145120.990874045]: euclidean_cluster > Setting points node to /points_raw
[ INFO] [1560145120.991440749]: Euclidean Clustering: Difference of Normals will not be used.
[ INFO] [1560145120.992293991]: [euclidean_clustering] downsample_cloud: 0
[ INFO] [1560145120.993129659]: [euclidean_clustering] remove_ground: 1
[ INFO] [1560145120.993895622]: [euclidean_clustering] leaf_size: 0.100000
[ INFO] [1560145120.994682790]: [euclidean_clustering] cluster_size_min 20
[ INFO] [1560145120.995661076]: [euclidean_clustering] cluster_size_max: 100000
[ INFO] [1560145120.996494191]: [euclidean_clustering] pose_estimation: 1
[ INFO] [1560145120.997390627]: [euclidean_clustering] clip_min_height: -1.300000
[ INFO] [1560145120.998214669]: [euclidean_clustering] clip_max_height: 0.500000
[ INFO] [1560145120.998939904]: [euclidean_clustering] keep_lanes: 0
[ INFO] [1560145120.999891451]: [euclidean_clustering] keep_lane_left_distance: 5.000000
[ INFO] [1560145121.000811750]: [euclidean_clustering] keep_lane_right_distance: 5.000000
[ INFO] [1560145121.001601722]: [euclidean_clustering] max_boundingbox_side: 10.000000
[ INFO] [1560145121.002341787]: [euclidean_clustering] cluster_merge_threshold: 1.500000
[ INFO] [1560145121.003157812]: [euclidean_clustering] output_frame: ego_vehicle/lidar/lidar1
[ INFO] [1560145121.003981865]: [euclidean_clustering] remove_points_upto: 0.000000
[ INFO] [1560145121.004793876]: [euclidean_clustering] clustering_distance: 0.750000
[ INFO] [1560145121.005570515]: [euclidean_clustering] use_gpu: 0
[ INFO] [1560145121.006381467]: [euclidean_clustering] use_multiple_thres: 0
[ INFO] [1560145121.007138562]: [euclidean_clustering] clustering_distances: [0.5,1.1,1.6,2.1,2.6]
[ INFO] [1560145121.007944239]: [euclidean_clustering] clustering_ranges: [15,30,45,60]
`

AravindChandradoss, 2019-06-10:
Install simple_pid and try again.
`sudo apt-get install ros-<your-ros-distro>-pid`

joel-mb, 2020-10-08:
Closing the issue as the repo has been largely updated. Feel free to file a new issue if any of the problems persist.

closed 2020-10-08 by joel-mb