carla-simulator/carla-autoware #50: RuntimeError: rpc::rpc_error during call in function version

AravindChandradoss, 2019-05-23:
I got this run time error. Any idea on what am I missing?
```
[INFO] [1558636356.781753, 0.000000]: listening to server localhost:2000
[INFO] [1558636356.781993, 0.000000]: using vehicle filter: vehicle.toyota.prius*
Traceback (most recent call last):
  File "/home/autoware/carla-autoware/catkin_ws/src/ros-bridge/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py", line 258, in <module>
    main()
  File "/home/autoware/carla-autoware/catkin_ws/src/ros-bridge/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py", line 251, in main
    ego_vehicle.run()
  File "/home/autoware/carla-autoware/catkin_ws/src/ros-bridge/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py", line 233, in run
    self.world = client.get_world()
RuntimeError: rpc::rpc_error during call in function version
```
I ran carla from UnrealEditor. (Not from docker)
Autoware was build using colcon. (latest release)

Here is entire output
```
started core service [/rosout]
process[carla_manual_control_ego_vehicle-2]: started with pid [24935]
process[carla_ros_bridge-3]: started with pid [24936]
process[goal_relay-4]: started with pid [24937]
process[initialpose_relay-5]: started with pid [24945]
process[carla_ego_vehicle_ego_vehicle-6]: started with pid [24959]
process[carla_waypoint_publisher-7]: started with pid [24962]
process[carla_points_map_loader-8]: started with pid [24966]
process[world_to_map-9]: started with pid [24973]
process[map_to_mobility-10]: started with pid [24976]
process[egovehiclegnss_to_gps-11]: started with pid [24985]
process[egovehiclelidar_to_velodyne-12]: started with pid [25002]
process[egovehiclecamerafront_to_camera-13]: started with pid [25007]
process[points_relay-14]: started with pid [25017]
process[imag_relay-15]: started with pid [25025]
process[odometry_to_posestamped-16]: started with pid [25029]
process[vehiclecmd_to_ackermanndrive-17]: started with pid [25034]
process[carla_to_autoware_vehicle_status-18]: started with pid [25045]
process[carla_to_autoware_waypoints-19]: started with pid [25051]
process[carla_ackermann_control_ego_vehicle-20]: started with pid [25058]
process[ray_ground_filter-21]: started with pid [25060]
process[joint_state_publisher-22]: started with pid [25064]
process[robot_state_publisher-23]: started with pid [25075]
process[can_odometry-24]: started with pid [25079]
process[voxel_grid_filter-25]: started with pid [25088]
process[ndt_matching-26]: started with pid [25093]
process[lidar_euclidean_cluster_detect-27]: started with pid [25096]
process[detection/lidar_detector/cluster_detect_visualization_01-28]: started with pid [25100]
process[can_status_translator-29]: started with pid [25104]
process[pose_relay-30]: started with pid [25111]
process[vel_relay-31]: started with pid [25119]
process[waypoint_marker_publisher-32]: started with pid [25127]
process[lane_rule-33]: started with pid [25131]
process[lane_stop-34]: started with pid [25137]
process[lane_select-35]: started with pid [25143]
process[costmap_generator-36]: started with pid [25153]
/usr/local/lib/python2.7/dist-packages/simple_pid/PID.py:22: UserWarning: time.monotonic() not available in python < 3.3, using time.time() as fallback
  warnings.warn('time.monotonic() not available in python < 3.3, using time.time() as fallback')
process[astar_avoid-37]: started with pid [25164]
process[velocity_set-38]: started with pid [25169]
process[config_waypoint_follower_rostopic-39]: started with pid [25188]
process[pure_pursuit-40]: started with pid [25205]
process[twist_filter-41]: started with pid [25212]
process[twist_gate-42]: started with pid [25216]
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
... logging to /home/autoware/.ros/log/21800e9a-7d89-11e9-a0c4-78d004236ed8/roslaunch-adl-Nuvo-5000-24966.log
[ INFO] [1558636356.298618714]: Initializing Ground Filter, please wait...
[ INFO] [1558636356.375458375]: Input point_topic: /points_raw
[ INFO] [1558636356.376912660]: [visualize_detected_objects] objects_src_topic: /detection/lidar_detector/objects
[ INFO] [1558636356.417324627]: [visualize_detected_objects] object_speed_threshold: 0.10
[ INFO] [1558636356.442487021]: sensor_height[meters]: 2.400000
[... 59 lines cut ...]
[ INFO] [1558636362.628208733]: [euclidean_clustering] downsample_cloud: 0
[ INFO] [1558636362.629025807]: [euclidean_clustering] remove_ground: 1
[ INFO] [1558636362.629961175]: [euclidean_clustering] leaf_size: 0.100000
[ INFO] [1558636362.630778136]: [euclidean_clustering] cluster_size_min 20
[ INFO] [1558636362.631564279]: [euclidean_clustering] cluster_size_max: 100000
[ INFO] [1558636362.632329344]: [euclidean_clustering] pose_estimation: 1
[ INFO] [1558636362.633066682]: [euclidean_clustering] clip_min_height: -1.300000
[ INFO] [1558636362.633871716]: [euclidean_clustering] clip_max_height: 0.500000
[ INFO] [1558636362.634954932]: [euclidean_clustering] keep_lanes: 0
[ INFO] [1558636362.635733841]: [euclidean_clustering] keep_lane_left_distance: 5.000000
[ INFO] [1558636362.636490109]: [euclidean_clustering] keep_lane_right_distance: 5.000000
[ INFO] [1558636362.637240591]: [euclidean_clustering] max_boundingbox_side: 10.000000
[ INFO] [1558636362.637990139]: [euclidean_clustering] cluster_merge_threshold: 1.500000
[ INFO] [1558636362.638884613]: [euclidean_clustering] output_frame: ego_vehicle/lidar/lidar1
[ INFO] [1558636362.639635739]: [euclidean_clustering] remove_points_upto: 0.000000
[ INFO] [1558636362.640409376]: [euclidean_clustering] clustering_distance: 0.750000
[ INFO] [1558636362.641123962]: [euclidean_clustering] use_gpu: 0
[ INFO] [1558636362.641888159]: [euclidean_clustering] use_multiple_thres: 0
[ INFO] [1558636362.642685150]: [euclidean_clustering] clustering_distances: [0.5,1.1,1.6,2.1,2.6]
[ INFO] [1558636362.643524764]: [euclidean_clustering] clustering_ranges: [15,30,45,60]
^C[twist_gate-42] killing on exit
[twist_filter-41] killing on exit
[pure_pursuit-40] killing on exit

```

closed 2019-05-30 by AravindChandradoss

osama-usuf, 2019-06-14:
Did you find a fix for this? Would be helpful if you could post it.

adi-geek2, 2019-08-02:
There was a way to stop getting RuntimeError: rpc::rpc_error by ensuring 

a.Same version of Python on all machines in n/w
b.Same version of CARLA on all machines in n/w
It worked for me, can someone please confirm that would be great ?

My setup
- Windows local (CARLA 0.9.5- server render machine) - Python 3.7.0
- Linux VM remote (CARLA 0.9.5- PythonAPI client manual_control.py) - Python 3.7.3

No4x, 2023-05-23:
I have solved the problem after installing the requirements from the examples.