carla-simulator/carla-autoware #51: Carla-autoware problem in setting 2D Nav Goal

StArbeiter, 2019-05-31:
Hi all,
I am running the carla-autoware demo using carla version 0.9.5, autoware version 1.11.0
I have started the carla server and autoware successfully and the manual control works fine in carla and visualization in Rviz is also fine. However, when I tried to set start/end of route, the problem showed up. 
Here is what I have done:
1. click '2D Pose Estimate', rviz shows 'Setting pose: -143.432 -27.764 -1.572 [frame=map]' and then the ego-vehicle respawned successfully.
2. click '2D Nav Goal', rviz shows 'Setting goal: Frame:map, Position(-109.006, -1.182, 0.000), Orientation(0.000, 0.000, 0.021, 1.000) = Angle: 0.041' and then nothing happened! Is there anything I haven't done? I checked the manual control mode in pygame, it says off. So help me if you know the problem in this demo.

Best

AravindChandradoss, 2019-05-31:
Try again. Sometimes it didn't work for me. Although I could see local/gobal waypoints in rviz.

StArbeiter, 2019-06-03:
> Try again. Sometimes it didn't work for me. Although I could see local/gobal waypoints in rviz.
> 
> Sometimes base-link I have issues with base_link. Let me know if you fix the issue.

I have tried 100+ times, and the ego-vehicle still not moving lol. So I am trying to find some details in source code. If you have solved this problem or used autoware successfully, let me know.

AravindChandradoss, 2019-06-03:
It was running fine. 
Were you able to see local/gobal waypoints in Rviz? 
The issue I had was; after sometime, base-link frame is behaving haphazardly. base-link tf was moving randomly although the lidar point were correctly localized.

StArbeiter, 2019-06-04:
No, i can not see the local/global waypoints in Rviz. 
Yes, I have the same issue, the base-link frame was moving haphazardly after I set the initial pose, and sometimes it was working okay.

AravindChandradoss, 2019-06-05:
did you use Autoware's rviz .config file?
In my case, I was able to see visualize both local/gobal waypoints. 
I still have base_link frame issue.

StArbeiter, 2019-06-12:
Yes, I have used autoware's rviz config file, and now I can see waypoints. However, the ego-vehicle is not moving after I set the goal, the number labeled on each waypoint is zero, can you make your ego_vehicle driving by setting the goal?