carla-simulator/carla-autoware #66: Autoware agent collides with obstacles

udeto, 2019-08-09:
Hi all,
I am running the carla-autoware bridge with CARLA 0.9.5 and autoware 1.12.0. The setup works and the agent follows the waypoint published, but it does not avoid obstacles on its route.

I am running autoware with lidar_euclidean_clustering, naive_motion_planning and costmap_generator for detection and astar_avoid, velocity_set and pure_pursuit for motion planning.

I expected the agent to detect the obstacles and calculate local waypoints accordingly to avoid a collision. Instead the agent sticks to its initial route even though the obstacles are detected by the lidar detector. 

Can somebody help me with this issue?

pprivulet, 2019-08-30:
Hi @udeto 
We have done the setup， but the car did not follow the path.  
Do you have the vector maps of the towns？
We are figuring out the problem，any suggests ？ thanks

joel-mb, 2020-10-08:
Hi @udeto @pprivulet 
The agent has been largely updated and now it does take into account obstacles.
The new vector maps can be found here: https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/

closed 2020-10-08 by joel-mb