carla-simulator/carla-autoware #74: Build success, run error:ResourceNotFound: waypoint_follower

OceanWong1991, 2019-09-11:
Hello everyone, 
 who encounter this problem : follow the build doc exectly, but when roslaunch $CARLA_AUTOWARE_ROOT/devel.launch
... logging to /home/ai/.ros/log/1e2b660a-d472-11e9-bc24-b06ebfc0c66c/roslaunch-DeepLearning-29621.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: waypoint_follower
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ai/carla-autoware/catkin_ws/src
ROS path [2]=/home/ai/autoware.ai/install/ymc/share
ROS path [3]=/home/ai/autoware.ai/install/xsens_driver/share
ROS path [4]=/home/ai/autoware.ai/install/wf_simulator/share
ROS path [5]=/home/ai/autoware.ai/install/lattice_planner/share
ROS path [6]=/home/ai/autoware.ai/install/waypoint_planner/share
ROS path [7]=/home/ai/autoware.ai/install/waypoint_maker/share
ROS path [8]=/home/ai/autoware.ai/install/way_planner/share
ROS path [9]=/home/ai/autoware.ai/install/vlg22c_cam/share
ROS path [10]=/home/ai/autoware.ai/install/vision_ssd_detect/share
ROS path [11]=/home/ai/autoware.ai/install/vision_segment_enet_detect/share
ROS path [12]=/home/ai/autoware.ai/install/vision_lane_detect/share
ROS path [13]=/home/ai/autoware.ai/install/vision_darknet_detect/share
ROS path [14]=/home/ai/autoware.ai/install/vision_beyond_track/share
ROS path [15]=/home/ai/autoware.ai/install/vehicle_socket/share
ROS path [16]=/home/ai/autoware.ai/install/vehicle_model/share
[... 67 lines cut ...]
ROS path [84]=/home/ai/autoware.ai/install/memsic_imu/share
ROS path [85]=/home/ai/autoware.ai/install/marker_downsampler/share
ROS path [86]=/home/ai/autoware.ai/install/map_tools/share
ROS path [87]=/home/ai/autoware.ai/install/map_tf_generator/share
ROS path [88]=/home/ai/autoware.ai/install/log_tools/share
ROS path [89]=/home/ai/autoware.ai/install/lidar_shape_estimation/share
ROS path [90]=/home/ai/autoware.ai/install/lidar_point_pillars/share
ROS path [91]=/home/ai/autoware.ai/install/lidar_naive_l_shape_detect/share
ROS path [92]=/home/ai/autoware.ai/install/lidar_fake_perception/share
ROS path [93]=/home/ai/autoware.ai/install/lidar_apollo_cnn_seg_detect/share
ROS path [94]=/home/ai/autoware.ai/install/libwaypoint_follower/share
ROS path [95]=/home/ai/autoware.ai/install/lgsvl_simulator_bridge/share
ROS path [96]=/home/ai/autoware.ai/install/lanelet2_validation/share
ROS path [97]=/home/ai/autoware.ai/install/lanelet2_examples/share
ROS path [98]=/home/ai/autoware.ai/install/lanelet2_python/share
ROS path [99]=/home/ai/autoware.ai/install/lanelet2_routing/share
ROS path [100]=/home/ai/autoware.ai/install/lanelet2_traffic_rules/share
ROS path [101]=/home/ai/autoware.ai/install/lanelet2_projection/share
ROS path [102]=/home/ai/autoware.ai/install/lanelet2_maps/share
ROS path [103]=/home/ai/autoware.ai/install/lanelet2_io/share
ROS path [104]=/home/ai/autoware.ai/install/lanelet2_core/share
ROS path [105]=/home/ai/autoware.ai/install/kvaser/share
ROS path [106]=/home/ai/autoware.ai/install/kitti_launch/share

THX, anyway

OceanWong1991, 2019-09-11:
Carla version : version 0.9.6
Autoware : ( 1.12.0)
@fpasch Sorry to bother you, I can't find any clue from the Doc or issues

OceanWong1991, 2019-09-11:
@fpasch if you need more information, just ask 

echo $ROS_PACKAGE_PATH

/home/ai/carla-autoware/catkin_ws/src:/home/ai/autoware.ai/install/ymc/share:/home/ai/autoware.ai/install/xsens_driver/share:/home/ai/autoware.ai/install/wf_simulator/share:/home/ai/autoware.ai/install/lattice_planner/share:/home/ai/autoware.ai/install/waypoint_planner/share:/home/ai/autoware.ai/install/waypoint_maker/share:/home/ai/autoware.ai/install/way_planner/share:/home/ai/autoware.ai/install/vlg22c_cam/share:/home/ai/autoware.ai/install/vision_ssd_detect/share:/home/ai/autoware.ai/install/vision_segment_enet_detect/share:/home/ai/autoware.ai/install/vision_lane_detect/share:/home/ai/autoware.ai/install/vision_darknet_detect/share:/home/ai/autoware.ai/install/vision_beyond_track/share:/home/ai/autoware.ai/install/vehicle_socket/share:/home/ai/autoware.ai/install/vehicle_model/share:/home/ai/autoware.ai/install/vehicle_gazebo_simulation_launcher/share:/home/ai/autoware.ai/install/vehicle_gazebo_simulation_interface/share:/home/ai/autoware.ai/install/vehicle_engage_panel/share:/home/ai/autoware.ai/install/vehicle_description/share:/home/ai/autoware.ai/install/trafficlight_recognizer/share:/home/ai/autoware.ai/install/op_utilities/share:/home/ai/autoware.ai/install/op_simulation_package/share:/home/ai/autoware.ai/install/op_local_planner/share:/home/ai/autoware.ai/install/op_global_planner/share:/home/ai/autoware.ai/install/lidar_kf_contour_track/share:/home/ai/autoware.ai/install/op_ros_helpers/share:/home/ai/autoware.ai/install/ff_waypoint_follower/share:/home/ai/autoware.ai/install/dp_planner/share:/home/ai/autoware.ai/install/op_simu/share:/home/ai/autoware.ai/install/op_planner/share:/home/ai/autoware.ai/install/op_utility/share:/home/ai/autoware.ai/install/lidar_euclidean_cluster_detect/share:/home/ai/autoware.ai/install/vector_map_server/share:/home/ai/autoware.ai/install/road_occupancy_processor/share:/home/ai/autoware.ai/install/costmap_generator/share:/home/ai/autoware.ai/install/object_map/share:/home/ai/autoware.ai/install/naive_motion_predict/share:/home/ai/autoware.ai/install/map_file/share:/home/ai/autoware.ai/install/libvectormap/share:/home/ai/autoware.ai/install/lane_planner/share:/home/ai/autoware.ai/install/imm_ukf_pda_track/share:/home/ai/autoware.ai/install/decision_maker/share:/home/ai/autoware.ai/install/vector_map/share:/home/ai/autoware.ai/install/vectacam/share:/home/ai/autoware.ai/install/udon_socket/share:/home/ai/autoware.ai/install/twist_generator/share:/home/ai/autoware.ai/install/twist_gate/share:/home/ai/autoware.ai/install/twist_filter/share:/home/ai/autoware.ai/install/tablet_socket/share:/home/ai/autoware.ai/install/state_machine_lib/share:/home/ai/autoware.ai/install/sound_player/share:/home/ai/autoware.ai/install/sick_lms5xx/share:/home/ai/autoware.ai/install/sick_ldmrs_tools/share:/home/ai/autoware.ai/install/sick_ldmrs_driver/share:/home/ai/autoware.ai/install/sick_ldmrs_msgs/share:/home/ai/autoware.ai/install/sick_ldmrs_description/share:/home/ai/autoware.ai/install/runtime_manager/share:/home/ai/autoware.ai/install/points2image/share:/home/ai/autoware.ai/install/rosinterface/share:/home/ai/autoware.ai/install/rosbag_controller/share:/home/ai/autoware.ai/install/roi_object_filter/share:/home/ai/autoware.ai/install/range_vision_fusion/share:/home/ai/autoware.ai/install/mpc_follower/share:/home/ai/autoware.ai/install/qpoases_vendor/share:/home/ai/autoware.ai/install/pure_pursuit/share:/home/ai/autoware.ai/install/pos_db/share:/home/ai/autoware.ai/install/points_preprocessor/share:/home/ai/autoware.ai/install/points_downsampler/share:/home/ai/autoware.ai/install/pixel_cloud_fusion/share:/home/ai/autoware.ai/install/lidar_localizer/share:/home/ai/autoware.ai/install/pcl_omp_registration/share:/home/ai/autoware.ai/install/pc2_downsampler/share:/home/ai/autoware.ai/install/oculus_socket/share:/home/ai/autoware.ai/install/obj_db/share:/home/ai/autoware.ai/install/nmea_navsat/share:/home/ai/autoware.ai/install/ndt_tku/share:/home/ai/autoware.ai/install/ndt_gpu/share:/home/ai/autoware.ai/install/ndt_cpu/share:/home/ai/autoware.ai/install/multi_lidar_calibrator/share:/home/ai/autoware.ai/install/mrt_cmake_modules/share:/home/ai/autoware.ai/install/mqtt_socket/share:/home/ai/autoware.ai/install/microstrain_driver/share:/home/ai/autoware.ai/install/memsic_imu/share:/home/ai/autoware.ai/install/marker_downsampler/share:/home/ai/autoware.ai/install/map_tools/share:/home/ai/autoware.ai/install/map_tf_generator/share:/home/ai/autoware.ai/install/log_tools/share:/home/ai/autoware.ai/install/lidar_shape_estimation/share:/home/ai/autoware.ai/install/lidar_point_pillars/share:/home/ai/autoware.ai/install/lidar_naive_l_shape_detect/share:/home/ai/autoware.ai/install/lidar_fake_perception/share:/home/ai/autoware.ai/install/lidar_apollo_cnn_seg_detect/share:/home/ai/autoware.ai/install/libwaypoint_follower/share:/home/ai/autoware.ai/install/lgsvl_simulator_bridge/share:/home/ai/autoware.ai/install/lanelet2_validation/share:/home/ai/autoware.ai/install/lanelet2_examples/share:/home/ai/autoware.ai/install/lanelet2_python/share:/home/ai/autoware.ai/install/lanelet2_routing/share:/home/ai/autoware.ai/install/lanelet2_traffic_rules/share:/home/ai/autoware.ai/install/lanelet2_projection/share:/home/ai/autoware.ai/install/lanelet2_maps/share:/home/ai/autoware.ai/install/lanelet2_io/share:/home/ai/autoware.ai/install/lanelet2_core/share:/home/ai/autoware.ai/install/kvaser/share:/home/ai/autoware.ai/install/kitti [... cut ...] _driveworks_gmsl_interface/share:/home/ai/autoware.ai/install/autoware_connector/share:/home/ai/autoware.ai/install/autoware_camera_lidar_calibrator/share:/home/ai/autoware.ai/install/autoware_build_flags/share:/home/ai/autoware.ai/install/autoware_bag_tools/share:/home/ai/autoware.ai/install/astar_search/share:/home/ai/autoware.ai/install/as/share:/home/ai/autoware.ai/install/amathutils_lib/share:/home/ai/autoware.ai/install/adi_driver/share:/opt/ros/kinetic/share

Can't find waypoint_follower?

closed 2019-09-23 by OceanWong1991

arashroomi, 2019-10-02:
I have same problem. Any updates on it?

SjorsGit, 2019-11-21:
I'm exploring Autoware with LG-simulator and I have the same problem when starting 'Motion Planning' via the runtime-manager:

Resource not found: waypoint_follower
ROS path [0]=/opt/ros/melodic/share/ros
etc.

In the quick launch example file ("my_motion_planning.launch") I had to change the lines that were searching for the "waypoint_follower" package:

  
  <node pkg="rostopic" type="rostopic" name="config_waypoint_follower_rostopic"
        args="pub -l /config/waypoint_follower autoware_config_msgs/ConfigWaypointFollower
        '{ header: auto, param_flag: 1, velocity: 14, lookahead_distance: 4.2, lookahead_ratio: 2.0, minimum_lookahead_distance: 6.0, displacement_threshold: 0.0, relative_angle_threshold: 0.0 }' "
  />
  ****
  <include file="$(find pure_pursuit)/launch/pure_pursuit.launch"/>

  
  ****
  <include file="$(find twist_filter)/launch/twist_filter.launch"/>

timoderbeste, 2020-04-06:
I have the same issue as well.

wuxiaohua1011, 2020-07-16:
Yeah, me too, how did you resolve this issue?

OceanWong1991, 2020-07-27:
> Yeah, me too, how did you resolve this issue?

Try recommand version