carla-simulator/carla-autoware #75: Setting up rviz

ishaan95, 2020-02-12:
Hello,

I've set up the project according to the instructions in the README. However I've not been able to find a launch script which creates an rviz node with the correct configuration. I tried launching 'carla ros bridge with rviz' from within the carla autoware but rviz doesn't show any data. 

Can I get a guideline on how to setup rviz with this project?

Thanks,
Ishaan

ishaan95, 2020-02-12:
I was able to run the visualizer by launching it from the autoware runtime manager. ([Link](https://gitlab.com/autowarefoundation/autoware.ai/autoware/-/wikis/ROSBAG-Demo)). However, the car still doesn't move. I'm stuck at the same point as ([This issue](https://github.com/carla-simulator/carla-autoware/issues/51)).

fpasch, 2020-02-12:
For visualization, please use the rviz-configurations that autoware provides or simply add all the topics manually.

Please keep in mind. In this repo only the bridging functionality and some example launch files are provided. Those launch-files + Autoware 1.12.0 should lead to a ego vehicle, that follows waypoints.
If this is not working, I would suggest, that you use rqt_graph+rostopic to analyze the data flow in order to identify the issue.

ishaan95, 2020-02-12:
Thank you for the response. I was able to run rviz from the runtime manager. I'm assuming that the default rviz config for autoware was used. However, the ego-vehicle is not moving. I have a some questions regarding this - 

1. Should the ego-vehicle be autonomously driving by default (right after running devel.launch)? 
2. After setting nav goal using rviz as shown on the README (which is not working for me right now), should we run the roslaunch script (devel.launch) again since this is how the waypoint publisher launch script is called.

I'm getting the following error from the waypoint pubisher python script (while running devel.launch)-
exit_waypoint=self._graph.edges[next_segment[0], next_segment[1]]['entry_waypoint'] with the error message as - TypeError: 'instancemethod' object has no attribute '__getitem__'
How do I check if all the waypoints are loaded correctly and if the car will move after running the launch script? 

My current status is very similar to #76 and #51. It would be great if a guideline is added to the README with details about this. 

Thank you,
Ishaan

jS5t3r, 2020-02-18:
I used the `.default`  from the quickstart tutorial.

KimHyung, 2020-04-21:
pip install simple-pid and retry it.