carla-simulator/carla-autoware #76: global waypoints

jS5t3r, 2020-02-12:
When launching `roslaunch $CARLA_AUTOWARE_ROOT/devel.launch`
and RVIZ I can not set the navigation goal. (I use ros melodic, autoware 1.12 and ubuntu 18.)

In rviz a expand the `Global Waypoints`:

```
global_velocity_lane_1/0 For frame [map]: No transform to fixed frame [velodyne].  TF error: [Lookup would require extrapolation into the future.  Requested time 138126.064153380 but the latest data is at time 138036.874152051, when looking up transform from frame [map] to frame [velodyne]]
global_velocity_lane_1/1 ...
global_velocity_lane_1/2 ...
global_velocity_lane_1/3 ...
```
How to fix this?

dmatos2012, 2020-02-16:
@jS5t3r maybe not an issue per se, but dont know where else to get information about this. I saw you have ros melodic, 1.12 and carla 0.97 on ubuntu 18.04, but it says carla 0.9.6 and kinetic required. Did you change the dockerfile or have you installed any other way(like build from source)? I am currently struggling so your help would be massive ot connect autoware + carla with ubuntu 18.04 and melodic(so your current setup).

jS5t3r, 2020-02-18:
@dmatos2012  on the discord channel I wrote with one of the carla developers and "there should not be any major problems."

1. CUDA **10.0** must be installed. In the `.zshrc` I added: 
```
export PATH=/usr/local/cuda-10.0/bin${PATH:+:${PATH}}$ 
export LD_LIBRARY_PATH=/usr/local/cuda-10.0/lib64${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
```
2. I installed from source version v1.12.0: https://gitlab.com/autowarefoundation/autoware.ai/autoware/-/wikis/Source-Build  

You can also use: (https://github.com/carla-simulator/carla-autoware/tree/master/docker)

I added these lines to my ~/.zshrc
```
source /opt/ros/melodic/setup.zsh
source ~/autoware.ai/install/setup.zsh
export CARLA_AUTOWARE_ROOT=~/carla-autoware/autoware_launch
export CARLA_MAPS_PATH=~/carla-autoware/autoware_data/maps
source $CARLA_AUTOWARE_ROOT/../catkin_ws/devel/setup.zsh
export PYTHONPATH=$PYTHONPATH:~/carla-python/carla/dist/carla-0.9.7-py2.7-linux-x86_64.egg:~/carla-python/carla/
```

jS5t3r, 2020-02-19:
@dmatos2012  could you manage to get something run? 

You have to change the role name from hero to ego_vehicle:
`roslaunch $CARLA_AUTOWARE_ROOT/devel.launch role_name"=ego_vehicle`

But it does not work all the time. I sometime I click select a new 2D Pose Estimate 20 times... 
The global waypoints are shown randomly, but if they are shown (fixed frame: world), I can calculate some routes...

@felipecode will this carla-autoware repository updated?  I saw that for Autoware1.14 is planned to have carla bridge inside: https://gitlab.com/vimalrajliya/simulation/-/tree/68afbc8a68ba13fc7fa67ce21e567c6943068ce7/carla_simulator_bridge

fpasch, 2020-02-19:
@jS5t3r Our plan is to move the content of this repo into autoware.

dmatos2012, 2020-02-19:
@jS5t3r Thanks man, thats a huge help! I have not been able to get it working yet, because we are still deciding what final setup we are gonna use. But at least i know its doable with your setup. and with Autoware 1.14!.

Let us know if you change something else to get it fully working.

jS5t3r, 2020-02-26:
I needed to download these https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/
into ~/carla-autoware/autoware_data/maps then it was fully working...

closed 2020-02-26 by jS5t3r