carla-simulator/carla-autoware #77: setting up with docker [still not working]

jS5t3r, 2020-02-17:
My setup:

Carla
```
#download docker image (e.g. version 0.9.6)
docker pull carlasim/carla:0.9.6

#extract the Carla Python API from the image
cd ~
mkdir carla-python
docker run --rm --entrypoint tar carlasim/carla:0.9.6 cC /home/carla/PythonAPI . | tar xvC ~/carla-python
```

Carla Autoware Bridge
```
cd ~
git lfs clone https://github.com/carla-simulator/carla-autoware.git
cd carla-autoware
git submodule update --init

cd docker
./build.sh
```
Run

```
Terminal 1:
nvidia-docker run -p 2000-2001:2000-2001 -it --rm carlasim/carla:0.9.6 ./CarlaUE4.sh

Terminal 2:
<in docker directory> ./run.sh
autoware@lh:~/Autoware$  roslaunch $CARLA_AUTOWARE_ROOT/devel.launch

Terminal 3:
<in docker directory> ./run.sh
https://gitlab.com/autowarefoundation/autoware.ai/autoware/-/wikis/ROSBAG-Demo
Launch RViz through the RViz button in the bottom-right corner of the ARM and load the default.rviz config provided with Autoware. To do this got to File -> Open Config and navigate to autoware.ai/src/autoware/documentation/autoware_quickstart_examples/launch/rosbag_demo/default.rviz
```

When I start, ` roslaunch $CARLA_AUTOWARE_ROOT/devel.launch`, I get the following error:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
Failed to find match for field 'intensity'.
Failed to find match for field 'intensity'.
[ndt_matching-26] process has died [pid 5809, exit code -6, cmd /home/autoware/Autoware/install/lidar_localizer/lib/lidar_localizer/ndt_matching __name:=ndt_matching __log:=/home/autoware/.ros/log/0b4e2286-5175-11ea-a5ab-00d861527c29/ndt_matching-26.log].
log file: /home/autoware/.ros/log/0b4e2286-5175-11ea-a5ab-00d861527c29/ndt_matching-26*.log
Failed to find match for field 'intensity'.
Failed to find match for field 'intensity'.
```

BTW:

sudo apt install ros-kinetic-desktop-full  in the autoware docker fixed the joint_state_publisher error

jasonxzzhang, 2020-02-27:
@jS5t3r Hi did you manage to get it to work? I am doing it the same way as you are doing, 

- running 'roslaunch $CARLA_AUTOWARE_ROOT/devel.launch' results in a CARLA ROS manual control window and a stationary vehicle, it does not respond to keyboard control.

- running 'rviz' on a separate terminal does not show anything linked to carla or the vehicle

Felix-Feng-90, 2020-06-08:
Hi, I have met the same problem, have you managed to find a solution for the runtime_error. BTW, the joint_state_publisher error can also be solved to install that package alone, it seems it is missing from the ros share directory.