carla-simulator/carla-autoware #78: Git LFS issue does not allow to checkout to old commit

brunoalr, 2020-02-26:
Hello, I want to test an older version of carla-autoware (specifically commit 7b71fba05be482458cf8cd95d92f68150448f01a) but when I call `git checkout 7b71fba05be482458cf8cd95d92f68150448f01a` I get the following error

```bash
git checkout 7b71fba05be482458cf8cd95d92f68150448f01a
Downloading autoware_data/map/map.pcd (127 MB)
Error downloading object: autoware_data/map/map.pcd (1ffe294): Smudge error: Error downloading autoware_data/map/map.pcd (1ffe294e6b4da8a342cf1dce1b0d7733db4e3e5a712440baf70e0e9e64c62a72): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to [suppressed]
Use `git lfs logs last` to view the log.
error: external filter git-lfs smudge -- %f failed 2
error: external filter git-lfs smudge -- %f failed
fatal: autoware_data/map/map.pcd: smudge filter lfs failed
```

Is there any workaround to this issue?

jS5t3r, 2020-02-26:
I downloaded it manually: https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/

brunoalr, 2020-02-26:
> I downloaded it manually: https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/

Do the new maps work with a previous bridge version?

jS5t3r, 2020-02-28:
I just have it from there: https://github.com/carla-simulator/carla-autoware/commit/bca5e0d79b1b4ff1681bbc003c42396264b44323 

I have not had the motivation to start an older version... But why just dont download them and try them out?