carla-simulator/carla-autoware #84: rviz_op_config.rviz

bsamadi, 2020-04-08:
I am following the steps in this [presentation](https://drive.google.com/file/d/1tTBMXOifFdjpgMEmM9Qjn7lO0nYpPs3A/view). I was wondering where I can find rviz_op_config.rviz.

DarrenWong, 2020-04-30:
if you are using carla 0.9.8, you can try to follow the latest bridge in here, https://gitlab.com/autowarefoundation/autoware.ai/simulation/-/tree/master/carla_simulator_bridge
It should be easier to integrate, but I still need some time to see how to let vehicle move.
[image]

jS5t3r, 2020-07-03:
@DarrenWong  Do you know what is the simplest way that the ego_vehicle is moving?  After following this tutorial (https://github.com/Autoware-AI/simulation/blob/master/carla_simulator_bridge/README.md) I found out that there is no (pure pursuit or sth). I guess with the runtime_manager I can do something meaningful:

```
roslaunch runtime_manager runtime_manager.launch
```

jS5t3r, 2020-07-03:
@bsamadi He mad a custom one.  https://github.com/Autoware-AI/autoware.ai/wiki/ROSBAG-Demo You can find one: `autoware.ai/src/autoware/documentation/autoware_quickstart_examples/launch/rosbag_demo/`

DarrenWong, 2020-07-03:
@jS5t3r  not yet, only move the car with Carla API with ros directly.

jS5t3r, 2020-07-08:
@DarrenWong  so, you mean manual control?

DarrenWong, 2020-07-08:
> @DarrenWong so, you mean manual control?

yes, manual control with ros bridge only... I am not further testing about it.... pls let me know if you succeed

wuxiaohua1011, 2020-07-16:
@DarrenWong , are you able to move the vehicle using Autoware? 

I am currently using Autoware 1.14 + Carla 0.9.9. 

When I have everything set up (So I see this screen), and everything, from Localization to waypoint, etc works
[image]

However, I am unable to move the vehicle because it seems like the launch file specified [here](https://github.com/carla-simulator/carla-autoware/blob/master/autoware_launch/my_motion_planning.launch#L67), find waypoint_follower, does not exist in Autoware 1.14 and 1.13. 

I do find it in 1.12, are you using 1.12?

DarrenWong, 2020-07-16:
not yet, seems you find a possible reason (not waypoint_follower) why we can not move, might be we can add the waypoint_follower node. I will try again if I have spare time later...

joel-mb, 2020-07-16:
Hi @DarrenWong @wuxiaohua1011 ,
`waypoint_follower` package has been removed in the newer versions of Autoware. You should replace this package with the newer ones (i.e., `pure_pursuit` and `twist_filter`). Hopefully, this will be solved in the new agent we are building (you can find more info here: https://github.com/carla-simulator/carla-autoware/tree/joel-mb/agent)

DarrenWong, 2020-07-17:
@joel-mb  your idea integrate all into docker is good. I am testing but I think the dockerfile have something not update yet

for example: 
1ssue 1:
` unused args [use_manual_control] for include of [/home/autoware/Autoware/install/carla_autoware_bridge/share/carla_autoware_bridge/launch/carla_autoware_bridge.launch]`
https://github.com/carla-simulator/carla-autoware/blob/55ce708e4023ff788a7e8be4630f701519c7a45e/carla-autoware-agent/launch/carla_autoware_agent.launch#L42

<arg name='use_manual_control' default='false'/>
The master of autoware-ai should have this one, but the docker file only build on top of the 

https://github.com/carla-simulator/carla-autoware/blob/55ce708e4023ff788a7e8be4630f701519c7a45e/Dockerfile#L2

https://github.com/Autoware-AI/simulation/blob/2677e815161e268171ec65a7543d14fb430bc9af/carla_simulator_bridge/carla_autoware_bridge/launch/carla_autoware_bridge.launch#L8

Tag: 1.14.0
https://github.com/Autoware-AI/simulation/blob/1.14.0/carla_simulator_bridge/carla_autoware_bridge/launch/carla_autoware_bridge.launch

Try to fix this, but still many errors after that.

Did you test the docker based on the autoware 1.14.0-melodic-cuda or the master version?

joel-mb, 2020-07-17:
Hi @DarrenWong!
Yes, you are right. I've done some modifications to the Dockerfile to align the different versions between carla and autoware. Also updated the `autoware-contents` submodule to point to the proper branch. I'm still working on some issues I've detected but it should work now.

When running the agent for the first time, you will get an error due to the map change. The issues comes from the `carla-ros-bridge`, so I'm taking a look to solve it.

Thanks for the feedback!

DarrenWong, 2020-07-18:
finally it moves after a few months trying, thanks a lot. But seems a small issue that we need to add the readme to get the map file because  it's a git lfs repo when try to pull the pcd 
https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/point_cloud_maps/Town01.pcd

[image]

And indeed a few issues like cannot select second goal, my yolo process died. But I believe it would be fix soonly

joel-mb, 2020-07-20:
Hi @DarrenWong!

*  Regarding selecting a second goal, unfortunately, this is a limitation introduced by the `open planner` package. As far as I know, you can select multiple goals at the beginning but not when the agent has started to drive. You may want to try with `lane_navi` node of the `lane_planner` package.
* Could you please provide more information about the issue with Yolo3? It shouldn't be crashing. Please, check that the `autoware-contents` submodule is correctly cloned and it includes the configuration files for yolo.
* The ERROR logs of CARLA are a known issue (https://github.com/carla-simulator/carla/issues/1821). We are working on it.

Thanks again for the feedback!

DarrenWong, 2020-07-20:
okay, the second question is I forget to pull all conetents in autoware-contents. New it works fine with modules. 
@wuxiaohua1011  @jS5t3r you can try the last version should be ok for functional test.
[image]

joel-mb, 2020-10-08:
Closing the issue. Feel free to re-open it if any of the problems persist.

closed 2020-10-08 by joel-mb

laizpp, 2021-04-23:
s

> ValueError: cannot reshape array of size 45579 into shape (11394,4)
> Traceback (most recent call last):
>   File "/home/autoware/carla_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/sensor.py", line 105, in _callback_sensor_data
>     self.sensor_data_updated(carla_sensor_data)
>   File "/home/autoware/carla_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/lidar.py", line 66, in sensor_data_updated
>     lidar_data, (int(lidar_data.shape[0] / 4), 4))
>   File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 232, in reshape
>     return _wrapfunc(a, 'reshape', newshape, order=order)
>   File "/usr/lib/python2.7/dist-packages/numpy/core/fromnumeric.py", line 57, in _wrapfunc
>     return getattr(obj, method)(*args, **kwds)
> ValueError: cannot reshape array of size 45579 into shape (11394,4)

Hi.How did you solve it?

sayfulloh11, 2023-02-27:
Hello!
Currently, are you using Carla 0.9.9 version?