carla-simulator/carla-autoware #88: waypoint_follower alternative/compatibility in Autoware 1.13+

wuxiaohua1011, 2020-07-16:
Hi,
I am using Carla 0.9.9 with Autoware 1.14. 

I am able to get everything setup and arrive at this screen 
[image]
However, my vehicle does not move no matter I used the[ carla's ROS node for publishing waypoint](https://github.com/carla-simulator/ros-bridge/tree/master/carla_waypoint_publisher) or I used the tool on RVIS. 

My vehicle does move in manual control mode. 

After looking into the issue, I found that I am missing the waypoint_follower node that is specified in the[ launch file](https://github.com/carla-simulator/carla-autoware/blob/master/autoware_launch/my_motion_planning.launch#L67). 

I went to the Autoware Gitlab and searched for it, it seems like that file is only in 1.12.

Can the maintainer make sure that this file can be in the newer versions of autoware or the launch file correspond to the correct version of autoware?

wuxiaohua1011, 2020-07-16:
Yeah, and I have confirmed that once you downgrade Autoware to 1.12, and you source the `autoware.ai/install/local_setup.bash`, and then you clone one of the earlier commits (I did this [one](https://github.com/carla-simulator/carla-autoware/tree/eec9d6a211f3e2cdc8790705186816427ba42669)

Follow the instruction on that commit for launching the carla-autoware bridge 
``` roslaunch $CARLA_AUTOWARE_ROOT/devel.launch ```

I have everything working, the vehicle is moving.

joel-mb, 2020-07-16:
Hi @wuxiaohua1011! Thanks for reporting the issue and the feeback.
You are right, the `waypoint_follower` package has been removed in the newer versions of Autoware and split into two different packages: `pure_pursuit` and `twist_filter` (If you replace the names in the launch file, everything should work as expected in Autoware 1.14).

Unfortunately, this repo has not been maintained until now. We are now updating it to make it work with the last version of Autoware, providing vector maps for Town01 and Town02 and a new agent (you can find more info in this branch: https://github.com/carla-simulator/carla-autoware/tree/joel-mb/agent).

closed 2020-10-08 by joel-mb