carla-simulator/carla-autoware #92: chroot: cannot change root directory to '/': Operation not permitted

wuxiaohua1011, 2020-07-30:
Hi,
I've been trying to run this new updated version. However, I encountering the issue of
`chroot: cannot change root directory to '/': Operation not permitted`

Here's a screenshot of what i did:
[image]

I am wondering how should i go about solving this issue? 

Thanks!

wuxiaohua1011, 2020-07-30:
Actually, I looked into this issue a bit, and it seems like this stack overflow says that you actually cannot start a docker at the root directory or something like that
[https://stackoverflow.com/questions/40636618/how-to-start-a-chrooted-directory-as-a-docker-container](https://stackoverflow.com/questions/40636618/how-to-start-a-chrooted-directory-as-a-docker-container)

Which means that the docker file might have a small problem?

charles-lei, 2020-08-04:
I hage the same problem.

twada-cats, 2020-08-05:
I also have same problem when run.sh is executed. I changed user as root before execution. 
$ sudo su
$ ./run.sh
It did not works. Can you tell us how to solve it?

mrm-mmh, 2020-08-11:
I've been able to find a workaround by adding sudo rights to the user in the run.sh and launching with sudo:
```
#!/bin/bash

XSOCK=/tmp/.X11-unix
XAUTH=$HOME/.Xauthority

VOLUMES="--volume=$XSOCK:$XSOCK:rw
         --volume=$XAUTH:$XAUTH:rw
         --volume=$(pwd)/autoware-contents:/home/autoware/autoware-contents:ro"

RUNTIME=""
DOCKER_VERSION=$(docker version --format '{{.Client.Version}}' | cut --delimiter=. --fields=1,2)
if [[ $DOCKER_VERSION < "19.03" ]] && ! type nvidia-docker; then
    RUNTIME="--gpus all"
else
    RUNTIME="--runtime=nvidia"
fi

docker run \
    -it --rm \
    $VOLUMES \
    --env="XAUTHORITY=${XAUTH}" \
    --env="DISPLAY=${DISPLAY}" \
    --privileged \
    --net=host \
    --user $(id -u ${USER}):$(id -g ${USER}) \
    --group-add sudo \
    $RUNTIME \
    carla-autoware:latest
```
`sudo ./run.sh` 

However, I agree with wuxiaohua1011 that the container should be adapted so that sudo is not required.

Hope that helps for now!

wuxiaohua1011, 2020-08-13:
> I've been able to find a workaround by adding sudo rights to the user in the run.sh and launching with sudo:
>
> ```
> #!/bin/bash
>
> XSOCK=/tmp/.X11-unix
> XAUTH=$HOME/.Xauthority [...]

Nice, it works great, thank you so much!

joel-mb, 2020-10-08:
Hi! The issue should be fixed now (https://github.com/carla-simulator/carla-autoware/pull/103). Feel free to re-open the issue if needed.

closed 2020-10-08 by joel-mb