carla-simulator/carla-autoware #95: build fails

saurabhjha1, 2020-08-16:
I am getting the following error message:
```
pathspec 'joel-mb/timeout' did not match any files(s) known to git
```

On deeper inspection, I believe the code is trying to checkout `joel-mb/timeout' but that branch does not exist. 

Should I checkout a stable branch/tag from 'carla-autoware' ? If yes, which one is the stable branch/tag?

ll7, 2020-08-18:
I have the same issue.
I thought it had something to do with the ros-bridge.
```Cloning into 'ros-bridge'...
Submodule 'carla_msgs' (https://github.com/carla-simulator/ros-carla-msgs) registered for path 'carla_msgs'
Cloning into '/home/autoware/ros-bridge/carla_msgs'...
Submodule path 'carla_msgs': checked out 'c89458b34aefbe7e12cefe3241c3b48f4a9b35d7'
error: pathspec 'joel-mb/timeout' did not match any file(s) known to git.
The command '/bin/bash -o pipefail -c git clone --recurse-submodules https://github.com/carla-simulator/ros-bridge.git     && cd ros-bridge && git checkout joel-mb/timeout' returned a non-zero code: 1
```

closed 2020-08-19 by joel-mb

joel-mb, 2020-08-19:
Hi @saurabhjha1 @ll7 !
I've just merged a PR fixing this. Thank you for pointing out the issue!

We've just detected another issue with the latest carla image. I hope this will be fixed in the next days. Meanwhile, I've created a new branch (`joel-mb/package_0_9_10`) which is using the PythonAPI from the package to overcome the problem.

ll7, 2020-08-19:
Thank you. So a `git pull` should fix #95? 

Does your temporary fix work with the latest carla docker image?

joel-mb, 2020-08-20:
Hi @ll7 ,
Yes, `git pull` should fix the issue with the ros-bridge branch. However, you'll find another issue with the latest carla docker image. This issue is not related to the carla-autoware integration. As soon as the carla docker image gets fixed everything will work as expected without any further modification.

Meanwhile, I've created a new branch called `joel-mb/package_0_9_10` to avoid using the latest carla docker image and use instead the latest package available (i.e., `0.9.10`). This solution is temporary until we fix the carla docker image. To use this branch just do the following:

```sh
git fetch
git checkout joel-mb/package_0_9_10
```
And follow the README document of this branch. Be aware that I've assumed that the user is using Ubuntu 18.04.

Hope this helps!

PR #97 (2020-08-19): Joel mb/ros bridge branch
Closes #95

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla-autoware/97)