carla-simulator/carla-autoware #99: some error about import carla, and cannot install libxerces in docker

gaohongfein, 2020-08-27:
File "/home/autoware/carla_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 24, in <module>
    import carla
  File "build/bdist.linux-x86_64/egg/carla/__init__.py", line 8, in <module>
    
  File "build/bdist.linux-x86_64/egg/carla/libcarla.py", line 7, in <module>
  File "build/bdist.linux-x86_64/egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: libxerces-c-3.1.so: cannot open shared object file: No such file or directory

joel-mb, 2020-08-28:
Hi @gaohongfein!
We are working on fixing this issue. It is not related to the autoware integration but to the CARLA docker image. Meanwhile, you can use the branch `joel-mb/package_0_9_10`.
Thanks!

joel-mb, 2020-10-08:
Hi @gaohongfein ! Please, follow the new instructions of the README. The issue should be gone. Feel free to re-open the issue if needed.

closed 2020-10-08 by joel-mb

TahaCarla, 2024-06-07:
Hey, I am still facing the same issue:
process[carla_ros_bridge-1]: started with pid [311]
process[goal_relay_ego_vehicle-2]: started with pid [312]
process[initialpose_relay_ego_vehicle-3]: started with pid [313]
process[carla_ego_vehicle_ego_vehicle-4]: started with pid [319]
process[carla_ackermann_control_ego_vehicle-5]: started with pid [325]
Traceback (most recent call last):
  File "/home/autoware/carla_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 28, in <module>
    import carla
ImportError: No module named carla
process[dynparam_docker_desktop_296_9153945959720019583-6]: started with pid [328]
process[world_to_map-7]: started with pid [331]