carla-simulator/carla-autoware #100: "Failed transform from base_link to ego_vehicle/lidar/lidar1" and autoware does not seem to connect properly

AIasd, 2020-09-09:
Hi,

I followed the setup and run the agent instructions of the branch `joel-mb/package_0_9_10` and modified `run.sh` as discussed in #92 . 

Everything seems fine until when the rviz window pops up. A screenshot is shown below: 

[image]

The only error message that ever appeared in the docker window is:
`[ERROR] [1599623780.456602301, 52.439418315]: Failed transform from base_link to ego_vehicle/lidar/lidar1`

Could you give me some hints on what goes wrong here?

AminoACID123, 2020-10-02:
Also experiencing this problem, hoping to find the answer.

joel-mb, 2020-10-08:
Hi @AIasd @AminoACID123. Can you try again following the new README instructions?

Basavaraj-PN, 2020-10-08:
I am getting the same issue

AminoACID123, 2020-10-09:
@joel-mb I used the updated Dockerfile to generate the docker image and everything works fine for me now. Thanks.

Jasonzhang11111, 2020-10-26:
Hi I am having the same problems, and the ego_vehicle is not moving at all. Could you please post solution if you managed to get it working? thanks

Basavaraj-PN, 2020-10-26:
> @joel-mb I used the updated Dockerfile to generate the docker image and everything works fine for me now. Thanks.

Still the same :(

Felix-Feng-90, 2020-10-28:
Not sure if it fits everyone, but for my case, I forget to install git lfs. Just need to install that, and redo the process. Then the problem is gone.

AminoACID123, 2020-10-29:
Based on my experiences, here is my advice:
1. Use the newest Dockerfile.
2. Check the integrity of the `autoware-contents`folder.
3. If it still doesn't work, check the log files located in `~/.ros/log/`folder for detailed information.

Jasonzhang11111, 2020-10-29:
thanks, I just got it working. I followed a comment from Felix-Feng-90 (can't find that comment anymore in here), basically I forgot to install Git lfs (silly mistake). After installing it, I repeated the whole process to build carla-autoware docker, alongside carla 0.9.10.1 docker, and everything works well for me

Felix-Feng-90, 2020-10-29:
@Jasonzhang11111 Glad to hear that installing Git lfs also works for you. I removed the comment because I found that although it works, there are still plenty of warning messages (especially some "topics not subscribed ones"), was trying to fix them as well. However, at the moment, I only solved the "state publisher" is deprecated warning, which needs to change the type of robot_state_publisher inside vehicle_model.launch

Basavaraj-PN, 2020-11-02:
Still the same with git-lfs

7n42Oaq1, 2020-11-04:
> thanks, I just got it working. I followed a comment from Felix-Feng-90 (can't find that comment anymore in here), basically I forgot to install Git lfs (silly mistake). After installing it, I [...]

Following the instruction I can view point clouds and the image in YOLO box. However, I still had the error Failed transform from base_link to ego_vehicle/lidar/lidar1. And the ego_vehicle did not move anymore even though I set the 2D Nav goal. Does someone encounter the problem and solve it?

joel-mb, 2020-11-04:
Hi @Basavaraj-PN @7n42Oaq1 ,
Are you able to see the vector map and the point cloud map in Rviz? As others have said, this may be related to not having the maps correctly cloned. Please check that inside the `autoware-contents` folder you have all the `*.pcd` , `*csv` and `*.osm` files.

Apart from this, can you check if there is any node crashing when running the agent?

Basavaraj-PN, 2020-11-04:
@joel-mb  I have attached terminal output and autoware-contents, 

[launch_log.txt](https://github.com/carla-simulator/carla-autoware/files/5488863/launch_log.txt)
[image]

7n42Oaq1, 2020-11-05:
@joel-mb my Rviz is shown like this
[image]
I think everything is ok except the vehicle does not move.

7n42Oaq1, 2020-11-05:
> @joel-mb my Rviz is shown like this
> [image]
> I think everything is ok except the vehicle does not move.

> @joel-mb my Rviz is shown like this
> [image]
> I think everything is ok except the vehicle does not move.

@joel-mb I found my problem that I modified the Dockerfile to compile ros packages in ros-bridge. I repeated the whole process to build carla-autoware docker follow the readme and now everything works. Thanks.

7n42Oaq1, 2020-11-05:
> @joel-mb my Rviz is shown like this
> [image]
> I think everything is ok except the vehicle does not move.

@joel-mb  I found my problem that I modified the Dockerfile to compile ros packages in ros-bridge. I repeated the whole process to build carla-autoware docker follow the readme and now everything works. Thanks.

joel-mb, 2020-11-06:
@7n42Oaq1 glad to here is working now for you!

@Basavaraj-PN I realized that the `vision_darknet_detect` node is also dying. Can you check what happends if you disable the vision nodes? Commenting the `vision_detection` include tag in `my_detection.launch` should be enough.

KailinTong, 2020-12-23:
Hello all,

I installed git-lfs and repeated the whole process again. 

Be patient when doing git clone, it might hang a while.

Then it works perfectly!

AI-tang, 2021-06-22:
I installed git-lfs and follow the read me to build the docker . but still get the error "Failed transform from base_link to ego_vehicle/lidar/lidar1", the ego vehicle still can not move

ghostsun89, 2021-09-03:
> I didn't use Docker to build the env and met the same issue. All is right unless the vehicle didn't move. I got the warning below:
> 
> ```
> [ WARN] [1630580298.268371780, 99.252712106]: Waiting for current_pose topic ...
> [ WARN] [1630580298.268480414, 99.252712106]: Waiting for current_velocity topic ...
> ```

```
pip install simple-pid pygame networkx==2.2
```

This fixed my issue.

110168511, 2021-12-08:
I installed git-lfs and repeated the whole process again.it is success.
thank you @KailinTong

kaka-lin, 2022-01-05:
Hi, @110168511 

I have the same issue. Could you please teach me how to fix this issue?
I already installed git-lfs but still not working

kaka-lin, 2022-01-06:
> Hello all,
> 
> I installed git-lfs and repeated the whole process again.
> 
> Be patient when doing git clone, it might hang a while.
> 
> Then it works perfectly!

Thank you @KailinTong !!!

lijecaru, 2022-03-14:
Hi, can someone please tell me the step-by-step process to run Carla and autoware docker? I did the following (Ubuntu20.04):
1. Installed Docker 19.03
2. Installed Nvidia Docker
3. Installed Carla Docker (0.9.10.1)
4. Installed Git-lfs (and checked that it was working with `$ git lfs install`)
5. Cloned Carla Autoware Repository `$ git clone --recurse-submodules https://github.com/carla-simulator/carla-autoware `
6. Built Carla autoware image to generate a `carla-autoware:latest `docker image
7. Run on one terminal the Carla server
8. Run the carla-autoware image
9. Start the agent by using: `$ roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town03 `
10. Rviz is showing the vehicle on one part of the map and the sensing suite in another. Camera is not working.
[image]

Error appears to be vehicle model, `No transform from [base_link] to [world] `

I did have to use sudo to run the autoware docker image: `$ sudo ./run.sh`

minhthai1995, 2022-03-16:
> 

Hi, which cuda version you are using?

zengtianle, 2025-04-07:
> Hi, can someone please tell me the step-by-step process to run Carla and autoware docker? I did the following (Ubuntu20.04):
>
> 1. Installed Docker 19.03
> 2. Installed Nvidia Docker
> 3. Installed Carla [...]

> Hi, can someone please tell me the step-by-step process to run Carla and autoware docker? I did the following (Ubuntu20.04):
>
> 1. Installed Docker 19.03
> 2. Installed Nvidia Docker
> 3. Installed Carla [...]

I have the same problem as you. Have you solved it? @lijecaru