carla-simulator/carla-autoware #102: chroot: cannot change root directory to '/': Operation not permitted

yumianhuli2, 2020-10-07:
Hello！

After  build the image with the following command:

cd carla-autoware && ./build.sh

I got 
REPOSITORY          TAG                   IMAGE ID            CREATED             SIZE
carla-autoware      latest                609bb5968338        13 hours ago        11GB

But Whether I run. /run.sh or sudo Docker run -it carla -autoware:latest /bin/bash I get a prompt that chroot: cannot change root Directory to '/': Operation not permitted

How can I fix this bug?Thanks！

joel-mb, 2020-10-07:
Hi @yumianhuli2 !
Can you try again? I just merged a PR fixing this. Please, remove the `carla-autoware` image and create it again using the `build` script.

yumianhuli2, 2020-10-08:
Hello！Thank you for update!
But When I run build script, I often encounter the following GPG errors:
[image]

yumianhuli2, 2020-10-08:
> Hello！Thank you for update!
> But When I run build script, I often encounter the following GPG errors:
> [image]
> 
> How can I solve this problem?Thank you！

> Hi @yumianhuli2 !
> Can you try again? I just merged a PR fixing this. Please, remove the `carla-autoware` image and create it again using the `build` script.

Hello！Thank you for update!
But When I run build script, I often encounter the following GPG errors:
[image]

How can I solve this problem?Thank you！

yumianhuli2, 2020-10-08:
> Hi @yumianhuli2 !
> Can you try again? I just merged a PR fixing this. Please, remove the `carla-autoware` image and create it again using the `build` script.

Hello！Thank you for update!
But When I run build script, I often encounter the following GPG errors:
[image]

How can I solve this problem?Thank you！

joel-mb, 2020-10-08:
Hi @yumianhuli2. Is this happening to you always? I cannot reproduce the error.

yumianhuli2, 2020-10-08:
> Hi @yumianhuli2. Is this happening to you always? I cannot reproduce the error.

Yes,Only succeeded once, others failed dozens of times at least.

yumianhuli2, 2020-10-09:
> Hi @yumianhuli2. Is this happening to you always? I cannot reproduce the error.

It works today,Thank you.
By the way, After I ran roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01,I got below on server
side:
[image]

got below on autoware side:
[image]

joel-mb, 2020-10-20:
Hi @yumianhuli2! Are you running the CARLA server and the autoware docker image on the same machine? I think this is not related to the GPU memory.

Can you please send me the complete log by running the following command:
```
roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01 > log.txt
```
This will generate a `log.txt` inside the container. To copy the file to your host machine run:

```
docker cp <CONTAINER_ID>:/home/autoware/log.txt .
``` 
You can get the `<CONTAINER_ID>` by running `docker ps`

Thanks!

yumianhuli2, 2020-11-09:
> Hi @yumianhuli2! Are you running the CARLA server and the autoware docker image on the same machine? I think this is not related to the GPU memory.
>
> Can you please send me the complete log by [...]

[log.txt](https://github.com/carla-simulator/carla-autoware/files/5507466/log.txt)

PR #103 (2020-10-07): Fixed ackermann and chroot bugs
Fixes #102 https://github.com/carla-simulator/carla-autoware/issues/92

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla-autoware/103)