carla-simulator/carla-autoware #104: NDT Localization setup

AhmedARadwan, 2020-10-12:
Hi,

Thank you for your great work. I'm trying to run ndt_matching with the vehicle moving in Town03 and ndt_matching doesn't work as expected as the transformation between ``` /world ``` frame to ``` /map ``` isn't specified correctly.

So my question here is how to get the correct values for the transformation between ``` /world ``` frame to ``` /map ``` frame in terms of x, y, z, roll, pitch, and yaw?

Thanks in advance,

joel-mb, 2020-10-20:
Hi @AdamGleave !
Which process are you following to execute the `ndt_matching` node? Are you just running the agent provided in this repo (i.e., `roslaunch carla_autoware_agent carla_autoware_agent.launch`)? Is this happening only in Town03 or you can reproduce it in all the other towns?

Thanks!

AdamGleave, 2020-10-20:
I think you mean @AhmedARadwan?

joel-mb, 2020-10-20:
yeah @AdamGleave! Thanks for the correction and sorry for the mistake.

AhmedARadwan, 2020-10-20:
Hi @joel-mb 

The problem occurs with all towns as the transformation between ```/world``` and ```/map``` is zeros
``` <!-- world to map tf -->
<node pkg="tf"  type="static_transform_publisher" name="world_to_map" args="0 0 0 0 0 0 /world /map 10" /> ```

I think it shouldn't be zeros as the GNSS location published by CARLA doesn't match map locations. That means that ndt_matching wouldn't work properly using GNSS locations and the pre-recorded map as they won't match and this is the case that happens with me.

Thanks