carla-simulator/carla-autoware #105: fail to generate ego_vehicle

lz1159435992, 2020-10-16:
when I do `roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01 host:=10.3.147.140`, it goes wrong and `[carla_ego_vehicle_ego_vehicle-4] process has died [pid 13695, exit code 1, cmd /home/autoware/carla_ws/src/ros-bridge/carla_ego_vehicle/src/carla_ego_vehicle/carla_ego_vehicle.py __name:=carla_ego_vehicle_ego_vehicle __log:=/home/autoware/.ros/log/3b70b1fe-0c41-11eb-9010-ac1f6b8091ec/carla_ego_vehicle_ego_vehicle-4.log].
log file: /home/autoware/.ros/log/3b70b1fe-0c41-11eb-9010-ac1f6b8091ec/carla_ego_vehicle_ego_vehicle-4*.log
 `occured. Only map changed in the carla, there isn't a car generated in the road.
I want to get some suggestions to fix this.

joel-mb, 2020-10-20:
Hi @lz1159435992!
Can you send me the complete log by running the following command:
```
roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01 host:=10.3.147.140 > log.txt
```
This will generate a `log.txt` inside the container. To copy the file to your host machine run:

```
docker cp <CONTAINER_ID>:/home/autoware/log.txt .
``` 
You can get the `<CONTAINER_ID>` by running `docker ps`:

lz1159435992, 2020-10-22:
[log.txt](https://github.com/carla-simulator/carla-autoware/files/5420796/log.txt)
this is the log.

joel-mb, 2020-11-04:
Hi @lz1159435992 
Sorry for the delay. Can you run again the command and send me the log (the `stderr` was not redirected to the file): 
```
roslaunch carla_autoware_agent carla_autoware_agent.launch > log.txt 2>&1
```

lz1159435992, 2020-11-09:
Thank you very much. After I use the computer with the display, I have already fix this problem. Maybe this is the root cause.

closed 2020-11-09 by lz1159435992