carla-simulator/carla-autoware #107: Inbound TCP/IP connection failed

Jasonzhang11111, 2020-10-27:
Hi I have been installing dockers for both Carla 0.9.10.1 and carla-autoware in this repo, and running them in parallel. 

Previously I could see the map and access the topics for the ego using rviz after `roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01`, problem I encountered earlier is it "Failed transform from base_link to ego_vehicle/lidar/lidar1", and the ego vehicle nevers moved.

However I build the carla-autoware image again yesterday, and I got 'Inbound TXP/IP connection failed' error, in rviz there is no map showing and many of the topics for the ego are missing, please see screenshots below. Could anyone please guide me on how to properly build the carla-autoware image, and how to run carla 0.9.10.1 docker with this carla-autoware? thanks

[image]
[image]
[image]

Jasonzhang11111, 2020-10-29:
I solved this one! I didn't install git lfs. After installing it, repeat the whole process and everything works

closed 2020-10-29 by Jasonzhang11111

oth54477, 2020-11-19:
Hi. 
I had the same error as you.
Can you tell me in detail how you solved it?

mattdst1, 2021-03-21:
Hi any updates on this one? I'm getting the same issue and not sure how to move forwards

jasonxzzhang, 2021-03-22:
hi @oth54477 @mattdst1, I can't remember exactly what I did that got it work as I tried many different ways on different PCs. I think if you haven't tried maybe worth re-cloning both Carla and Autoware, and make sure that git lfs is installed

mattdst1, 2021-03-22:
Brilliant thank you for the information. I’ll give a re clone a go.

Kind regards
Matt