carla-simulator/carla-autoware #108: Weird behavior of agent

nikola-j, 2020-10-30:
Hi,

The agent just doesn't work after a couple of turns, it happens every time I run autoware
https://drive.google.com/file/d/17421n3pFJwFab53uSruxy9wzysnjIik1/view?usp=sharing
Did anyone else have this issue happen to him?

MTDzi, 2020-10-30:
I've seen a similar behavior in a very different setup and it had to do with the fact that the localization module was either lagging or stopped working altogether.

Would it be possible to check the output of `rostopic hz /name_of_the_localization_topic`?

joel-mb, 2020-11-06:
Hi @nikola-j! I've never seen a similar issue. Did you be able to check if any of the nodes were dying as @MTDzi suggested?