carla-simulator/carla-autoware #109: How to set new destinations for ego vehicle?

AminoACID123, 2020-11-05:
Right now I am able to get the car moving by setting `2d nav goal` in rviz. But it seems that I cannot set new destinations for the car, even when it's reached the current destination. Are there any suggestions for this?

joel-mb, 2020-11-06:
Hi @AminoACID123 ,
Unfortunately, this is a limitation introduced by the global planner of the `OpenPlanner` package. For more advanced behaviors I suggest you using the newest versions of this package (https://gitlab.com/hatem-darweesh-autoware.ai/core_planning/-/tree/openplanner.1.13). Please, take into account that the newest versions of the OpenPlanner is maintained by its creator but it's not part of the official Autoware repository.
Hope this helps!

AminoACID123, 2020-11-14:
I'll have a try. Thanks for your advice. 
By the way, what is the max detection distance of Autoware?

joel-mb, 2020-11-23:
@AminoACID123 I'm afraid that I cannot give you a concrete number. I think this is highly dependent on the configuration of your detection nodes and the configuration of the sensors. I suggest you taking a look to the different parameters Autoware accepts as well as trying different parameters for the CARLA sensors. You can take as a starting point the configuration of the agent provided in this repo:
- CARLA sensor configuration: https://github.com/carla-simulator/carla-autoware/blob/master/carla-autoware-agent/agent/config/sensors.json
- Autoware agent launch files: https://github.com/carla-simulator/carla-autoware/tree/master/carla-autoware-agent/agent/launch

AminoACID123, 2020-11-24:
I see. Thank you very much.

closed 2020-11-24 by AminoACID123