carla-simulator/carla-autoware #110: Carla simulator crashes while connecting

pirate-lofy, 2020-11-19:
I start the simulator on my host machine by this command 
`./CarlaUE4.sh -windowed -ResX=800 -ResY=600 -carla-server` 
after the simulator is up, I fire the launch file from the docker container by
`roslaunch carla_autoware_agent carla_autoware_agent.launch town:=Town01`
the simulator crashes. this is the full log of Carla
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 3762] 
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 281, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=128368 LargeMemoryPoolOffset=259504 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```

and this is the full log of the roslaunch 
```
... logging to /home/autoware/.ros/log/368092b6-2aa2-11eb-92f7-cc2f71e7786d/roslaunch-bignrz-Lenovo-ideapad-520-15IKB-822.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bignrz-Lenovo-ideapad-520-15IKB:36051/

SUMMARY
========

PARAMETERS
 * /astar_avoid/angle_goal_range: 6.0
 * /astar_avoid/avoid_start_velocity: 3.0
 * /astar_avoid/avoid_waypoints_velocity: 10.0
 * /astar_avoid/closest_search_size: 30
 * /astar_avoid/curve_weight: 1.2
 * /astar_avoid/distance_heuristic_weight: 1.0
 * /astar_avoid/enable_avoidance: False
 * /astar_avoid/lateral_goal_range: 0.5
 * /astar_avoid/longitudinal_goal_range: 2.0
 * /astar_avoid/minimum_turning_radius: 6.0
 * /astar_avoid/obstacle_threshold: 100
 * /astar_avoid/potential_weight: 10.0
 * /astar_avoid/replan_interval: 2.0
 * /astar_avoid/reverse_weight: 2.0
 * /astar_avoid/robot_base2back: 1.0
 * /astar_avoid/robot_length: 4.5
 * /astar_avoid/robot_width: 1.75
 * /astar_avoid/safety_waypoints_size: 100
 * /astar_avoid/search_waypoints_delta: 2
 * /astar_avoid/search_waypoints_size: 50
 * /astar_avoid/theta_size: 48
 * /astar_avoid/time_limit: 1000.0
 * /astar_avoid/update_rate: 10
 * /astar_avoid/use_back: False
 * /astar_avoid/use_potential_heuristic: True
 * /astar_avoid/use_wavefront_heuristic: False
 * /carla/ackermann_control/accel_Kd: 0.05
 * /carla/ackermann_control/accel_Ki: 0.0
 * /carla/ackermann_control/accel_Kp: 0.05
 * /carla/ackermann_control/min_accel: 1.0
 * /carla/ackermann_control/speed_Kd: 0.5
 * /carla/ackermann_control/speed_Ki: 0.0
 * /carla/ackermann_control/speed_Kp: 0.05
 * /carla/ego_vehicle/role_name: ['hero', 'ego_veh...
 * /carla/fixed_delta_seconds: 0.05
 * /carla/host: localhost
 * /carla/port: 2000
 * /carla/synchronous_mode: False
 * /carla/synchronous_mode_wait_for_vehicle_control_command: True
[... 400 lines cut ...]
[ INFO] [1605816722.714023585]: [visualize_detected_objects] model_color: {R:190.0, G:190.0, B:190.0, A:0.5}
[ INFO] [1605816722.714420008]: [visualize_detected_objects] centroid_color: {R:77.0, G:121.0, B:255.0, A:0.8}
[ INFO] [1605816722.716427199]: [visualize_detected_objects] object_src_topic: /detection/lidar_tracker/objects
[ INFO] [1605816722.717280787]: [visualize_detected_objects] markers_out_topic: /detection/lidar_tracker/objects_markers
================================================================================REQUIRED process [carla_ros_bridge-1] has died!
process has finished cleanly
log file: /home/autoware/.ros/log/368092b6-2aa2-11eb-92f7-cc2f71e7786d/carla_ros_bridge-1*.log
Initiating shutdown!
================================================================================
process[ukf_track_relay_01-34]: started with pid [1105]
RLException: cannot add process [vision_darknet_detect-35] after process monitor has been shut down
The traceback for the exception was written to the log file
[ukf_track_relay_01-34] killing on exit
[detection/lidar_tracker/ukf_track_visualization_01-33] killing on exit
[imm_ukf_pda_01-32] killing on exit
[detection/l_shaped/naive_shape_visualization_01-31] killing on exit
[lidar_naive_l_shape_detect-30] killing on exit
[detection/lidar_detector/cluster_detect_visualization_01-29] killing on exit
[detection/lidar_detector/object_roi_filter_clustering-28] killing on exit
[lidar_euclidean_cluster_detect-27] killing on exit
[ndt_matching-26] killing on exit
[voxel_grid_filter-25] killing on exit
autoware@bignrz-Lenovo-ideapad-520-15IKB:/home/autoware$ terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
```

joel-mb, 2020-11-23:
Hi @pirate-lofy,
One issue I've seen in your log is that the `ego_vehicle` node is not receiving the `world_info` topic on time. Can you try setting the map first and then running autoware.

1. Start the server:
```sh
./CarlaUE4.sh -windowed -ResX=800 -ResY=600 -carla-server
```
2. Change to Town01 (the `config.py` in under the `PythonAPI/util` folder).
```sh
python config.py -m Town01
```
3. Run the carla-autoware agent.

pirate-lofy, 2020-11-23:
Hi, @joel-mb , thanks for replying.
I tried to change the map many times using the config.py file but the simulator crashes after running the config.py file witth following error

```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 3762] 
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 268, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=128368 LargeMemoryPoolOffset=259504 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```

and then I tried to change the map using the terminal parameter 
`/Game/Carla/Maps/Town01`
but it doesn't change the map

joel-mb, 2020-11-24:
@pirate-lofy which CARLA package are you using? Are you using 0.9.10.1?

pirate-lofy, 2020-11-24:
Yes, I am using 0.9.10.1 pre-build release

xichennn, 2021-07-23:
same problem here, have you solved it?

SepShr, 2021-12-22:
Hi,
I am facing similar problem while trying to run Carla with [Pylot](https://github.com/erdos-project/pylot). Following, is the full log of my Carla run:
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
LowLevelFatalError [File:Unknown] [Line: 3762]
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 268, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=132912 LargeMemoryPoolOffset=264048
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
The failure happened when I ran Pylot and I also confirmed that Pylot was successfully running, until it shutdown since it was not able to connect to Carla. Interestingly, Carla did not consume any GPU memory which is really odd!

I appreciate any help, if possible. Thanks.