carla-simulator/carla-autoware #117: Make Carla Map usable for Autoware

MichaelW93, 2021-01-06:
Hello,

we have created a custom Map for Carla, which works just fine. Now we want to use autoware to drive our vehicle in the town.
I was already able to create a point cloud of the Town using the pcl_recorder.
We now want autoware to calculate its own path. For this, autoware requires a vector map of the town afaik 
Until now I was not able to find a way to convert our town into the proper format needed by autoware. 
So how do I create a vector map of our carla map for autoware, or how was it done with the already provided ones.
Much thanks in advance!

EsquilliDev, 2021-02-10:
Hey @MichaelW93, This week I had to do the same thing. My steps to get my custom map into Autoware were:

1.  RoadRunner - OpenDRIVE file (.xodr)
2.  [Assure Mapping Tool](https://github.com/hatem-darweesh/assuremappingtools) - Lanelet2 (.xodr to .osm)
3. [Autoware lanelet2aisan](https://gitlab.com/autowarefoundation/autoware.ai/utilities/-/tree/392a0465359d51f63fb877170a330912d71a0537/lanelet_aisan_converter) - Aisan vector map (.osm to .csv)
4. Place your map files into their respective folder inside Autoware's docker like this:
- PCD: /home/autoware/autoware-contents/maps/point_cloud_maps/MyCustomMap.pcd
- Aisan: /home/autoware/autoware-contents/maps/vector_maps/asian/MyCustomMap./<'all generated files'>.csv
- Lanelet2: /home/autoware/autoware-contents/maps/vector_maps/lanelet2/MyCustomMap.osm

5. Launch Autoware
`roslaunch carla_autoware_agent carla_autoware_agent.launch town:=MyCustomMap`

Notes:

-  Currently, CARLA's Autoware only load the PCD and Aisan maps.
-  I'm having an issue in which the PCD and the vector map doesn't match. They are misaligned.

athulg, 2021-02-15:
Hey @Esquilli , I'm new to this and I'm trying to convert .osm to .csv using lanelet2aisan. Can you please guide me on where to run the command(steps required) `rosrun lanelet_aisan_converter lanelet2aisan _map_file:=<path to lanelet map> _origin_lat:=<latitude of origin for XYZ projection> _origin_lon:=<longitude of origin for XYZ projection> _save_dir:=<optional: path to save dir>`

Thanks!

Esquilli, 2021-02-15:
@athulg You need to run that command inside the carla-autoware docker. You can run `roscore` outside the docker, but that commands still needs to be run inside the docker container.

athulg, 2021-02-15:
@Esquilli Thanks a lot!

Esquilli, 2021-02-15:
@athulg Let me know if after exporting the map and trying it on Autoware you get the same issue as me (mismatched/misaligned maps). Did you used Assure mapping tool?

athulg, 2021-02-15:
@Esquilli I'm sorry I actually don't remember how I got the .osm file. But, now I've a different issue after exporting it on Autoware (Unintialized quaternion, assuming identity) in Rviz

Esquilli, 2021-02-15:
I haven't experienced that issue myself, so I don't know how to help you on that one.