carla-simulator/carla-autoware #123: "No image in YOLO", "Necessary topics are not subscribed yet" and the ego car doesn't move.

slotDumpling, 2021-03-26:
[image]
[image]

Hi, I followed the README, and repeated for several times. I have git-lfs installed, and it did cloned the submodules.
How to solve this? Thank you!

closed 2021-03-27 by slotDumpling

xichennn, 2021-08-12:
@sparklin1999 same issue here. Have you found any solutions?

kaka-lin, 2022-01-05:
Hi, same issue here.

caseymeiz, 2023-03-13:
Same issue. Why was this closed?

AOOOOOA, 2023-04-16:
Same issues

xichennn, 2023-06-15:
The not subscribed topics may not be an issue. Try to give the ego car a
goal in Rviz (just click goal in navi bar then click anywhere on the map)
and see if it moves.