carla-simulator/carla-autoware #124: Step 14 of build.sh failed

alvinliu97, 2021-04-05:
Got this error running ./build.sh , i have alr tried update pip and python

```
Step 14/20 : RUN pip install simple-pid pygame networkx==2.2
 ---> Running in 2509aa2c3ba4
Collecting simple-pid
  Downloading https://files.pythonhosted.org/packages/f0/23/fa4e51987a031580f3af54a75eae60aacce593e23669075ae65664511244/simple_pid-1.0.0-py2.py3-none-any.whl
Collecting pygame
  Downloading https://files.pythonhosted.org/packages/d2/11/4212b97ea025aca55e2284e9c7d8cb87cfaddb4e89f5eb613c81017246c1/pygame-2.0.1-cp27-cp27mu-manylinux1_x86_64.whl (11.7MB)
Collecting networkx==2.2
  Downloading https://files.pythonhosted.org/packages/f3/f4/7e20ef40b118478191cec0b58c3192f822cace858c19505c7670961b76b2/networkx-2.2.zip (1.7MB)
Collecting decorator>=4.3.0 (from networkx==2.2)
  Downloading https://files.pythonhosted.org/packages/71/bd/042e63027bea950da4e6f5545eb59b973868da854ab30a18128d3b884104/decorator-5.0.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-nLxXse/decorator/setup.py", line 4, in <module>
        exec(open('src/decorator.py').read(), dic)  # extract the __version__
      File "<string>", line 162
        print('Error in generated code:', file=sys.stderr)
                                              ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-nLxXse/decorator/
The command '/bin/bash -o pipefail -c pip install simple-pid pygame networkx==2.2' returned a non-zero code: 1
```

kanakab, 2021-04-07:
I am also getting the same error, any help from the team would be great

alvinliu97, 2021-04-07:
> I am also getting the same error, any help from the team would be great

hi I made it work by installing pygame separately and hashtag the dockerfile step 14

kanakab, 2021-04-07:
@alvinliu97 by step 14, do you mean "RUN pip install simple-pid pygame networkx==2.2"

alvinliu97, 2021-04-07:
yes

kanakab, 2021-04-07:
thanks a lot, I think the issue is solved now

closed 2021-04-08 by alvinliu97