carla-simulator/carla-autoware #126: Carla Vector Map with Traffic Light

thiagodeborba, 2021-04-26:
Hello,

Can anyone provide me a Carla Vector Map with Traffic Light?
Can be from any Town, e.g. Town01.
It is only for research purpose.

Thank You!

liuypcoding, 2022-08-12:
> Hello,
> 
> Can anyone provide me a Carla Vector Map with Traffic Light? Can be from any Town, e.g. Town01. It is only for research purpose.
> 
> Thank You!

Hi,Had you get The Carla Vector&Point Map with Traffic Light? I need to .

THX!!

raozhongyu, 2024-02-21:
> > Hello,
> > Can anyone provide me a Carla Vector Map with Traffic Light? Can be from any Town, e.g. Town01. It is only for research purpose.
> > Thank You!
> 
> Hi,Had you get The Carla Vector&Point Map with Traffic Light? I need to .
> 
> THX!!
Hi, did you get the Carla Vector map with carla