carla-simulator/carla-autoware #131: Autoware warnings -- "base_link" passed to lookupTransform argument target_frame does not exist. Necessary topics are not subscribed yet ...

mattdst1, 2021-07-01:
I'm running Carla 0.9.10.1 using Carla Autoware. Previously ran and installed this image without issues. Running fresh install of Ubuntu 18.04 lts. Verified that autoware-contents all installed correctly. 

Can't figure out how to debug this issue and really need some help on this one. 

I'm getting the issues shown below in the terminal:

[ WARN] [1625137898.387848978, 856.744570121]: "base_link" passed to lookupTransform argument target_frame does not exist. 
[ERROR] [1625137898.387909340, 856.744570121]: Failed transform from base_link to ego_vehicle/lidar/lidar1
[ WARN] [1625137899.366378586, 857.744570136]: "base_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1625137899.394870991]: Necessary topics are not subscribed yet ... 
[ WARN] [1625137899.424981928, 857.794570136]: Necessary topics are not subscribed yet. Waiting...
[ WARN] [1625138423.215167301, 6124.144648611]: Necessary topics are not subscribed yet ... 
[ INFO] [1625138423.244008353, 6124.144648611]: [range_vision_fusion] Empty Detections. Make sure the vision and range detectors are running.

AOOOOOA, 2023-04-20:
> I'm running Carla 0.9.10.1 using Carla Autoware. Previously ran and installed this image without issues. Running fresh install of Ubuntu 18.04 lts. Verified that autoware-contents all installed correctly.
> 
> Can't figure out how to debug this issue and really need some help on this one.
> 
> I'm getting the issues shown below in the terminal:
> 
> * [ WARN] [1625137899.394870991]: Necessary topics are not subscribed yet ...
> * [ WARN] [1625137899.424981928, 857.794570136]: Necessary topics are not subscribed yet. Waiting...
> * [ WARN] [1625137898.387848978, 856.744570121]: "base_link" passed to lookupTransform argument target_frame does not exist.
> * [ERROR] [1625137898.387909340, 856.744570121]: Failed transform from base_link to ego_vehicle/lidar/lidar1
> 
> Then either:
> 
> * [ INFO] [1625138423.244008353, 6124.144648611]: [range_vision_fusion] Empty Detections. Make sure the vision and range detectors are running.
> 
> Or:
> 
> [roi_object_filter] Empty input messages, for details check the topics : /grid_map_wayarea and /detection/lidar_detector/objects.
> 
> [image] [image]

Hi, I also met this problem. How do you solve it?  thanks

mattdst1, 2023-04-20:
Nope sorry, ended up leaving the PhD looking at this.

closed 2024-10-12 by mattdst1