carla-simulator/carla-autoware #140: carla-autoware don't support carla version 0.9.13

SandyWare, 2022-03-30:
**Environment** : 
Ubuntu 18.0.4
Carla  version : 0.9.13
Ros : Melodic
Issue:  **carla-autoware (0.910) don't  support carla version 0.9.13**

Error Description : 
```
[rosout][INFO] 2022-03-30 05:36:33,776: Trying to connect to localhost:2000
[rosout][WARNING] 2022-03-30 05:36:33,782: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API. **Client API version: 0.9.10.1. Simulator API version: 0.9.13-1-g8854804f4**

```

zehranrgi, 2023-06-02:
It is just a warning, does it work or not?

CurlyWei, 2023-07-20:
The environment that  can work normally,I did
Environment :
Ubuntu 18.0.4
Carla version : 0.9.13
Ros : Melodic

CurlyWei, 2023-07-20:
The environment that can work normally,I did
But I use the docker to build.

Environment :
Local -
Ubuntu 20.0.4
Docker -
Ubuntu 18.0.4
Carla version : 0.9.13
Ros : Melodic