carla-simulator/carla-autoware #142: CARLA-Autoware localization problem

Jk7913, 2022-04-15:
Ubuntu 18.04
CARLA 0.9.11
Autoware 1.14

Hi, I'm currently using Carla 9.11 but same thing was happening with 9.10.1. It also does not mater if I'm running ROS-bridge in synchronous or asynchronous mode, problems is always persistent. Basically on sharp turn, when NDT time increase, Autoware get lost, looks like vehicle shown in rviz Autoware is not capable to keep up with the vehicle in simulation. I assume something is wrong with localization in Autoware. I'm still getting some errors/warnings but I do not know if it has to do anything with it ("_Failed transform from base_link to ego_vehicle/lidar_", and some others which might be related to that one), since when the vehicle is driving straight, whole thing looks ok. Any idea what might be causing the problem? Thanks

[image]

MakerFace, 2022-07-12:
I have the same problem!

MakerFace, 2022-07-12:
I'm not sure if it actually solved the problem, but you can try this solution:
Add the following code snippet
```xml
  <!--
    # imu #
    remap carla imu to autoware
    @todo: to reduce load, Autoware should directly use the Carla-topic.
  -->
  <node pkg='topic_tools' type='relay' name='imu_relay' args='/carla/$(arg role_name)/imu/imu1 /imu_raw'/>
```
to `carla_autoware/carla-autoware-agent/agent/bridge.launch`, and restart your carla_autoware_agent. Gook luck.