carla-simulator/carla-autoware #144: Carla-Autoware controlling ego-vehicle

tecena, 2022-06-16:
Hi ,
I am new to autoware. I have trying to a carla open scenario file with autoware,where the ego is completely controlled by autoware.
1.Carla 0.9.11 is set-up in a windows machine.
2.I have followed the https://github.com/carla-simulator/carla-autoware to setup carla-autoware setup in ubuntu.
3.This docker images pulls the ros bridge as well as carla-autoware-bridge.
4.When I run the cmd roslaunch carla_autoware_agent carla_autoware_agent.launch the carla-autoware is getting connected to carla simulator as I am able to fetch clock data from rostopics.
5.Now when I try to run a scenario and use autoware as a controller, I did start the FolloLeadingVehicle.xosc file in carla with scenariorunner and did run the launch file  roslaunch carla_autoware_agent carla_autoware_agent.launch , just vehicles in scenarios get spawned but the vehicle does not move at all and there are couple of errors and warnings on the autoware treminal
**[ INFO] [1655377342.896036185, 713.344599086]: DecisionMaker is waiting filtered_point for NDT
[ WARN] [1655377342.896326826, 713.344599086]: Necessary topics are not subscribed yet. Waiting...
[ WARN] [1655377342.896790692, 713.344599086]: Necessary topics are not subscribed yet ...
[ WARN] [1655377342.906116402, 713.394599087]: Necessary topics are not subscribed yet ...
[ WARN] [1655377342.906975138, 713.394599087]: Necessary topics are not subscribed yet ...
[ WARN] [1655377342.916686490, 713.444599088]: Necessary topics are not subscribed yet ...
[ERROR] [1655377342.925575422, 713.494599089]: Could not find a connection between 'velodyne' and 'map' because they are not part of the sametree.Tf has two or more unconnected trees.

I tried alot to find the solution , but unable to understand how to solve these errors and why ego vehicle is not moving in openscenario execution, 
Please help

ThiagoFelipeSandeiro, 2022-06-20:
Hi tecena,
Did you solve this issue? Have you tried setting the "2D Pose Estimation" but clicking the button in RVIZ then clicking and dragging in the direction you want the ego to go? Also, after setting the "2D Pose Estimation", do the same with "2D Nav Goal". Not sure if this helps but couldn't hurt throwing it out there...

YuqiHuai, 2022-06-24:
[image]
Same Issue here. I can set 2D Pose Estimation, I see point cloud visualization but the center isn't the vehicle showing in RVIZ.
Setting 2D Nav Goal gives me a path, but nothing moves/changes at all.

tecena, 2022-06-27:
> Hi tecena, Did you solve this issue? Have you tried setting the "2D Pose Estimation" but clicking the button in RVIZ then clicking and dragging in the direction you want the ego to go? Also, after [...]

Thank you for you reply
The autoware set-up is in linux machine which does not have any RDP or monitor set-up .I am using autoware only through the terminal. So can you please suggest how do I set the goal position from terminal with commands without clicking on the "2d Pose Estimation" and "2d Nav Goal"

ThiagoFelipeSandeiro, 2022-06-28:
> [image] Same Issue here. I can set 2D Pose Estimation, I see point cloud visualization but the center isn't the vehicle showing in RVIZ. Setting 2D Nav Goal gives me a path, but nothing moves/changes at all.

Hi YuqiHuai, 
Whick version of carla-autoware are you running? Are you getting any errors in the terminal that is running carla-autoware?
 

> Thank you for you reply The autoware set-up is in linux machine which does not have any RDP or monitor set-up .I am using autoware only through the terminal. So can you please suggest how do I set [...]

Hi tecena,
You can set the start pose and goal pose via a csv file called EgoCar.csv. In the csv you can set one or multiple goals. The format/header is 'X,Y,Z,A,C,V,name,' and is located here ~/autoware_openplanner_logs/SimulationData. Also, you need to change this line "<arg name='enableRvizInput' value='true' />"  in the my_mission_planning.launch file to false.
Browse this file to find out more... ~/autoware.ai/src/autoware/core_planning/op_global_planner/nodes/op_global_planner_core.cpp
I'm in the process of trying to implement this also. I have been getting many errors when I run the roslaunch command in the carla-autoware docker, thus, I'm in the process of updating the carla-autoware docker to which also requires me to update carla-simulator and ros-bridge.

YuqiHuai, 2022-06-28:
@ThiagoFelipeSandeiro I am not sure how I can get version information. I just cloned this repo and started.
I was having issue with docker set up so I added a few lines to the Dockerfile
```
RUN apt-key del 7fa2af80
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/3bf863cc.pub
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu2004/x86_64/7fa2af80.pub
```
I also saw
```
[pcl::PCDReader::readHeader] No points to read
load failed /home/autoware//autoware-contents/maps/point_cloud_maps/Town01.pcd
load /home/autoware//autoware-contents/maps/point_cloud_maps/Town01.pcd
```
and a lot of
```
[ WARN] [1656459901.681647637, 11.934444092]: Necessary topics are not subscribed yet ..
```

Maybe the missing pcd files are the cause?

tecena, 2022-06-29:
> Hi YuqiHuai, Whick version of carla-autoware are you running? Are you getting any errors in the terminal that is running carla-autoware?
>
>
> Hi tecena, You can set the start pose and goal pose via a [...]

@ThiagoFelipeSandeiro 

In my_mission_planning.launch file <arg name='enableRvizInput' value='true' />  I did change this value to false.

But I am not able to find the EgoCar.csv and the path/folder  ~/autoware_openplanner_logs/SimulationData is also not available. Am I missing any steps in execution or any more installations so that I can give goal point through the .csv file and run my Ego along the path.

ThiagoFelipeSandeiro, 2022-06-30:
Hi @tecena @YuqiHuai,
Okay so here's the update... 
I have updated; carla-simulator to version 0.9.11; carla-autoware to version 0.9.11; ros-bridge to version 0.9.11.
I found the autoware-contents is mounted to the docker from the local drive which is unreliable so copy it over the the docker, and then when in the running docker you can create the EgoCar.csv with echo commands. Then when you run the roslaunch command add the spawn-point arg with the same values as the start point in the EgoCaro.csv - note: the different data structure with the two coords - then when RVIZ pops up the ego should start automatically.
I have provided the commands and mod files here [https://github.com/ThiagoFelipeSandeiro/carla-autoware-mods](https://github.com/ThiagoFelipeSandeiro/carla-autoware-mods).
Hope this helps. It has worked for me.

tecena, 2022-06-30:
> Hi @tecena @YuqiHuai, Okay so here's the update... I have updated; carla-simulator to version 0.9.11; carla-autoware to version 0.9.11; ros-bridge to version 0.9.11. I found the autoware-contents is [...]

@ThiagoFelipeSandeiro  
Thank you for your guidance,
I will work on this and update you

YuqiHuai, 2022-06-30:
@ThiagoFelipeSandeiro Thanks for the information! I got busy with other stuff and will try this out over the weekend!

tecena, 2022-07-05:
> Hi @tecena @YuqiHuai, Okay so here's the update... I have updated; carla-simulator to version 0.9.11; carla-autoware to version 0.9.11; ros-bridge to version 0.9.11. I found the autoware-contents is [...]

Here you mention in the note  that  when the rviz pops up the ego should start automatically. Do we still need rviz even though we specify start and destination points in CSV.I do not need any visualization .
So can skip running the rviz part